osrf/ros2_test_cases #1097: The Keystore

marcoag, 2024-04-29:
Check the documentation for the 'The Keystore' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [The Keystore page](https://docs.ros.org/en/jazzy/Tutorials/Advanced/Security/The-Keystore.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-16:
@marcoag i can check this, sign me up.

fujitatomoya, 2024-05-16:
no problem observed, but minor text fix required https://github.com/ros2/ros2_documentation/pull/4463

<details>
  <summary>test result(click to expand!)</summary>

```shell
root@tomoyafujita:~/sros2_demo/demo_keystore/public# openssl x509 -in ca.cert.pem -text -noout
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            4e:f5:49:6a:ff:e1:01:be:07:1d:92:ab:d1:81:de:64:17:17:b8:97
        Signature Algorithm: ecdsa-with-SHA256
        Issuer: CN = sros2CA
        Validity
            Not Before: May 15 05:06:36 2024 GMT
            Not After : May 14 05:06:36 2034 GMT
        Subject: CN = sros2CA
        Subject Public Key Info:
            Public Key Algorithm: id-ecPublicKey
                Public-Key: (256 bit)
                pub:
                    04:86:48:be:39:6a:6d:fe:aa:7e:7e:2d:af:fe:87:
                    64:16:f8:a1:4b:80:80:a5:e2:d7:e9:30:05:54:4e:
                    c8:ea:7f:93:03:9a:ca:7d:2a:33:1e:6f:f2:9a:db:
                    8f:73:25:f3:49:8a:01:8d:7a:43:6d:14:3b:58:ce:
                    56:62:d4:3a:5f
                ASN1 OID: prime256v1
                NIST CURVE: P-256
        X509v3 extensions:
            X509v3 Basic Constraints: critical
                CA:TRUE, pathlen:1
    Signature Algorithm: ecdsa-with-SHA256
    Signature Value:
        30:45:02:21:00:b1:36:5b:18:f3:94:ef:5a:c3:cc:88:18:f6:
        8d:d4:e6:2d:c2:79:3d:fa:7b:4a:f0:0c:dc:0e:d6:11:94:99:
        43:02:20:6c:41:7b:6e:b2:2a:44:fe:ba:83:6e:ad:c6:21:2f:
        0d:8c:ea:80:98:b8:72:ef:c1:3e:eb:b3:fd:5c:a8:8e:48

root@tomoyafujita:~/sros2_demo/demo_keystore/enclaves# openssl smime -verify -in governance.p7s -CAfile ../public/permissions_ca.cert.pem
Content-Type: text/plain

<dds xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="http://www.omg.org/spec/DDS-SECURITY/20170901/omg_shared_ca_governance.xsd">
    <domain_access_rules>
        <domain_rule>
            <domains>
              <id>0</id>
            </domains>
            <allow_unauthenticated_participants>false</allow_unauthenticated_participants>
            <enable_join_access_control>true</enable_join_access_control>
            <discovery_protection_kind>ENCRYPT</discovery_protection_kind>
            <liveliness_protection_kind>ENCRYPT</liveliness_protection_kind>
            <rtps_protection_kind>SIGN</rtps_protection_kind>
            <topic_access_rules>
                <topic_rule>
                    <topic_expression>*</topic_expression>
                    <enable_discovery_protection>true</enable_discovery_protection>
                    <enable_liveliness_protection>true</enable_liveliness_protection>
                    <enable_read_access_control>true</enable_read_access_control>
                    <enable_write_access_control>true</enable_write_access_control>
                    <metadata_protection_kind>ENCRYPT</metadata_protection_kind>
                    <data_protection_kind>ENCRYPT</data_protection_kind>
                </topic_rule>
            </topic_access_rules>
        </domain_rule>
    </domain_access_rules>
</dds>
Verification successful
```

</details>