osrf/ros2_test_cases #1102: Installation Ubuntu

marcoag, 2024-04-29:
Check the documentation for the 'Installation Ubuntu' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Installation Ubuntu page](https://docs.ros.org/en/jazzy/Tutorials/Advanced/Simulators/Webots/Installation-Ubuntu.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

samuel-ongzx, 2024-05-14:
## Process
- [x] Step 1 - I was able to follow the documentation.
- [x] Step 2 - The documentation seemed clear to me.
- [x] Step 3 - The documentation didn't have any obvious errors.

No issues with documentation, it is clear and can be followed.

## Findings
Error arose from lack of `jazzy` support in webots. Also an error in `webots_controller`, see image below.
- binary: `http://packages.ros.org/ros2-testing/ubuntu noble InRelease`: only available packages in `jazzy` are `ros-jazzy-webots-ros2-importer`, `ros-jazzy-webots-ros2-msgs` and `ros-jazzy-webots-ros2-tests`. 
- src: [github](https://github.com/cyberbotics/webots_ros2/tree/master/webots_ros2_driver), lack `jazzy` support in `CMakeLists.txt`, particularly in [`webots_ros2_driver`](https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_driver/CMakeLists.txt). Modifying it to add `jazzy` allows python to be found and compiles successfully.

## Verification
- Able to launch `ros2 launch webots_ros2_universal_robot multirobot_launch.py`
[image]

- Got error in `webots_controller`
```
[INFO] [webots_controller_abbirb4600-4]: process started with pid [1021926]
[webots_controller_abbirb4600-4] Error:   No link elements found in urdf file
[webots_controller_abbirb4600-4]          at line 207 in ./urdf_parser/src/model.cpp
[webots_controller_abbirb4600-4] Failed to parse robot description using: urdf_xml_parser/URDFXMLParser
[webots_controller_abbirb4600-4] terminate called after throwing an instance of 'std::runtime_error'
[webots_controller_abbirb4600-4]   what():  URDF cannot be parsed by a `ros2_control` component parser:
[webots_controller_abbirb4600-4] [ros2run]: Aborted
[ERROR] [webots_controller_abbirb4600-4]: process has died [pid 1021926, exit code 250, cmd '/home/user/webots_ws/install/webots_ros2_driver/share/webots_ros2_driver/scripts/webots-controller --robot-name=abbirb4600 --protocol=ipc --port=1234 ros2 --ros-args -r __ns:=/abb -p robot_description:=/home/user/webots_ws/install/webots_ros2_universal_robot/share/webots_ros2_universal_robot/resource/webots_abb_description.urdf -p use_sim_time:=True --params-file /home/user/webots_ws/install/webots_ros2_universal_robot/share/webots_ros2_universal_robot/resource/ros2_control_abb_config.yaml'].
```

marcoag, 2024-06-26:
Reported the missing packages in the issue above. Thanks for the checks @samuel-ongzx!

closed 2024-06-26 by marcoag