osrf/ros2_test_cases #1103: Installation Windows

marcoag, 2024-04-29:
Check the documentation for the 'Installation Windows' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Installation Windows page](https://docs.ros.org/en/jazzy/Tutorials/Advanced/Simulators/Webots/Installation-Windows.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

Ryanf55, 2024-05-01:
I don't think this combo makes sense. You can't follow the windows instructions on Ubuntu.

clalancette, 2024-05-01:
> I don't think this combo makes sense. You can't follow the windows instructions on Ubuntu.

Yep, good point.  Closing this one.

closed 2024-05-01 by clalancette