osrf/ros2_test_cases #1110: Beginner CLI Tools

marcoag, 2024-04-29:
Check the documentation for the 'Beginner CLI Tools' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Beginner CLI Tools page](https://docs.ros.org/en/jazzy/Tutorials/Beginner-CLI-Tools.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

jasmeet0915, 2024-05-11:
All the links work correctly on the page!

clalancette, 2024-05-12:
Thanks for testing!

closed 2024-05-12 by clalancette