osrf/ros2_test_cases #1111: Configuring ROS2 Environment

marcoag, 2024-04-29:
Check the documentation for the 'Configuring ROS2 Environment' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Configuring ROS2 Environment page](https://docs.ros.org/en/jazzy/Tutorials/Beginner-CLI-Tools/Configuring-ROS2-Environment.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

ottojo, 2024-05-01:
I successfully tested this, here are my notes:

## Step 1 - I was able to follow the documentation.
* I followed the instructions for sourcing `/opt/ros/jazzy/setup.bash`
* Environment variables were set as expected
* Adding command to bashrc worked
* (not in tutorial, but sourcing workspace succeeded, verified with demo_nodes_cpp talker/listener)
* setting ROS_DOMAIN_ID succeeded (and worked, verified with same demo)
* same worked with ROS_DOMAIN_ID in bashrc
* ROS_AUTOMATIC_DISCOVERY_RANGE: as mentioned, discovery worked on local network by default
* (also verified that changing ROS_AUTOMATIC_DISCOVERY_RANGE changes this behavior, which is not covered here but in a linked tutorial)

-> Suggested steps all worked as described, output as expected

## Step 2 - The documentation seemed clear to me.

Environment setup is clear, only note would be that the introduction mentioned workspaces and overlays, which will probably only become clear once a workspace with source packages is used above the system packages, but i guess that is covered in later tutorials.

Expected pitfalls are explained clearly and solutions are offered, as with ROS_AUTOMATIC_DISCOVERY_RANGE

## Step 3 - The documentation didn't have any obvious errors.

No issues occurred during testing, no factual errors have been found while reading the page

marcoag, 2024-05-01:
Great, thanks for the testing!!

closed 2024-05-01 by marcoag