osrf/ros2_test_cases #1115: Understanding ROS2 Actions

marcoag, 2024-04-29:
Check the documentation for the 'Understanding ROS2 Actions' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Understanding ROS2 Actions page](https://docs.ros.org/en/jazzy/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Actions/Understanding-ROS2-Actions.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

ganatrask, 2024-05-02:
I tried this

I was able to follow the documentation.
The documentation seemed clear to me and didn't had any obvious error 

[image]

[image]

Can you assign this to me and close this issue?

Thanks.

marcoag, 2024-05-02:
Thanks for testing!

closed 2024-05-02 by marcoag