osrf/ros2_test_cases #1125: Custom ROS2 Interfaces

marcoag, 2024-04-29:
Check the documentation for the 'Custom ROS2 Interfaces' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Custom ROS2 Interfaces page](https://docs.ros.org/en/jazzy/Tutorials/Beginner-Client-Libraries/Custom-ROS2-Interfaces.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

Benjamin-Tan, 2024-05-06:
Able to follow the documentation, it is very clear, and no obvious errors. Every step and output is as expected.

clalancette, 2024-05-06:
Thanks for testing!

closed 2024-05-06 by clalancette