osrf/ros2_test_cases #1128: Single Package Define And Use Interface

marcoag, 2024-04-29:
Check the documentation for the 'Single Package Define And Use Interface' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Single Package Define And Use Interface page](https://docs.ros.org/en/jazzy/Tutorials/Beginner-Client-Libraries/Single-Package-Define-And-Use-Interface.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

Benjamin-Tan, 2024-05-06:
Able to follow the documentation, it is very clear, and no obvious errors. Every step and output is as expected. The extras are very useful

clalancette, 2024-05-06:
Thanks for testing!

closed 2024-05-06 by clalancette