osrf/ros2_test_cases #1129: Using Parameters In A Class CPP

marcoag, 2024-04-29:
Check the documentation for the 'Using Parameters In A Class CPP' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Using Parameters In A Class CPP page](https://docs.ros.org/en/jazzy/Tutorials/Beginner-Client-Libraries/Using-Parameters-In-A-Class-CPP.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

slowrunner, 2024-05-05:
Unless someone beats me to this one, I'd like to work this test later today.

slowrunner, 2024-05-05:
For the config:
- Chip set: AMD64
- Build type: Debian
- OS: Ubuntu Noble 24.04
- DDS: FastDDS

I followed the ["Using Params in a C++ Class"](https://docs.ros.org/en/jazzy/Tutorials/Beginner-Client-Libraries/Using-Parameters-In-A-Class-CPP.html) Tutorial

1) No problems following the doc as written
2) Doc was clear what to do and how
3) No errors detected

```
 ros2 launch cpp_parameters cpp_parameters_launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-05-05-14-17-29-756349-Ubuntu-Noble-FusionVM-12490
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [minimal_param_node-1]: process started with pid [12493]
[minimal_param_node-1] [INFO] [1714933050.880644073] [custom_minimal_param_node]: Hello earth!
[minimal_param_node-1] [INFO] [1714933051.880712610] [custom_minimal_param_node]: Hello world!
[minimal_param_node-1] [INFO] [1714933052.880534024] [custom_minimal_param_node]: Hello world!
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[minimal_param_node-1] [INFO] [1714933053.722250585] [rclcpp]: signal_handler(signum=2)
[INFO] [minimal_param_node-1]: process has finished cleanly [pid 12493]
```

# Again noting that Ctrl-C in C++ node examples is graciously handled - Why not so in Python?

closed 2024-05-06 by ahcorde