osrf/ros2_test_cases #1130: Using Parameters In A Class Python

marcoag, 2024-04-29:
Check the documentation for the 'Using Parameters In A Class Python' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Using Parameters In A Class Python page](https://docs.ros.org/en/jazzy/Tutorials/Beginner-Client-Libraries/Using-Parameters-In-A-Class-Python.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

slowrunner, 2024-05-05:
I started this one - getting an error - need to investigate:
```
$ ros2 run python_parameters minimal_param_node
Traceback (most recent call last):
  File "/home/ubuntu/nobledesk/ros2ws/install/python_parameters/lib/python_parameters/minimal_param_node", line 33, in <module>
    sys.exit(load_entry_point('python-parameters==0.0.0', 'console_scripts', 'minimal_param_node')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ubuntu/nobledesk/ros2ws/install/python_parameters/lib/python_parameters/minimal_param_node", line 25, in importlib_load_entry_point
    return next(matches).load()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/importlib/metadata/__init__.py", line 205, in load
    module = import_module(match.group('module'))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1324, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'python_parameters.python_parameters_node'
[ros2run]: Process exited with failure 1
```

slowrunner, 2024-05-05:
Found my error - file was named python_parameters_nod.py  (missing the final e)

This tutorial for the specified configuration:
1) I was able to follow to success
2) It was clear - even for debugging my problem
3) No errors found

clalancette, 2024-05-17:
I've now opened up https://github.com/ros2/rclpy/issues/1280 , which should make these kinds of things easier to fix in the future.  Given that, I'm actually going to call this "complete" and close it.

Thanks for testing!

closed 2024-05-17 by clalancette