osrf/ros2_test_cases #1133: Writing A Simple Py Publisher And Subscriber

marcoag, 2024-04-29:
Check the documentation for the 'Writing A Simple Py Publisher And Subscriber' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Writing A Simple Py Publisher And Subscriber page](https://docs.ros.org/en/jazzy/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Publisher-And-Subscriber.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

slowrunner, 2024-05-05:
For config:
- Chip set: AMD64
- Build type: Debian
- OS: Ubuntu Noble 24.04
- DDS: FastDDS

I followed the ["Writing A Simple Py Publisher And Subscriber" ](https://docs.ros.org/en/jazzy/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Publisher-And-Subscriber.html)

1) no problem following the doc
2) doc was clear - very nicely presented
3) doc did not have any errors that I could detect

Programs ran correctly:
```
$ ros2 run py_pubsub talker
[INFO] [1714923200.378470173] [minimal_publisher]: Publishing: "Hello World: 0"
[INFO] [1714923200.863956235] [minimal_publisher]: Publishing: "Hello World: 1"
[INFO] [1714923201.364496024] [minimal_publisher]: Publishing: "Hello World: 2"

$ ros2 run py_pubsub listener 
[INFO] [1714923221.881084123] [minimal_subscriber]: I heard: "Hello World: 43"
[INFO] [1714923222.364977761] [minimal_subscriber]: I heard: "Hello World: 44"

```

BUT - I have felt this tutorial has always suffered from not demonstrating a clean termination.  Control-C termination should not raise an "Exception" - it is a normal/expected event that a publisher and subscriber (any ROS 2 node actually) should be written to handle.

Pressing Ctrl-C generates a scary/nasty "you did something wrong":
```
[INFO] [1714923224.865000141] [minimal_subscriber]: I heard: "Hello World: 49"
^CTraceback (most recent call last):
  File "/home/ubuntu/nobledesk/ros2ws/install/py_pubsub/lib/py_pubsub/listener", line 33, in <module>
    sys.exit(load_entry_point('py-pubsub==0.0.0', 'console_scripts', 'listener')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ubuntu/nobledesk/ros2ws/install/py_pubsub/lib/python3.12/site-packages/py_pubsub/subscriber_member_function.py", line 41, in main
    rclpy.spin(minimal_subscriber)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/__init__.py", line 240, in spin
    executor.spin_once()
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 827, in spin_once
    self._spin_once_impl(timeout_sec)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 811, in _spin_once_impl
    handler, entity, node = self.wait_for_ready_callbacks(
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 781, in wait_for_ready_callbacks
    return next(self._cb_iter)
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 685, in _wait_for_ready_callbacks
    wait_set.wait(timeout_nsec)
KeyboardInterrupt
[ros2run]: Interrupt
```

with similar when stopping the publisher.

# Why do we scare NOOBs like this?
(While I have been messing with ROS/ROS 2 for years now, I have never progressed to be able to create a pub/sub with a timer to catch KeyboardInterrupt and shutdown cleanly.  I once asked on StackExchange and got the answer to use LifeCycle Nodes - perhaps it is just not possible in ROS 2 to code a clean Ctrl-C exit?)

clalancette, 2024-05-07:
> Why do we scare NOOBs like this?

This has to do with how Python handles Ctrl-C, though our code is also involved.  In short, Python will always throw this exception on Ctrl-C.  The best you can do as a user is to catch it and ignore it.  We probably should do that in our examples; maybe I'll spend some time tomorrow adding that.

marcoag, 2024-06-26:
Closing this as it's been more than one month since the tutorial party ended. Will add the Ctrl-C to the backlog of things to improve. Thanks @slowrunner!

closed 2024-06-26 by marcoag