osrf/ros2_test_cases #1136: Content Filtering Subscription

marcoag, 2024-04-29:
Check the documentation for the 'Content Filtering Subscription' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Content Filtering Subscription page](https://docs.ros.org/en/jazzy/Tutorials/Demos/Content-Filtering-Subscription.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-14:
@marcoag i can take care of this.

fujitatomoya, 2024-05-14:
No problem found, we can close this.

<details>
  <summary>test result(click to expand!)</summary>

```shell
root@tomoyafujita:~# ros2 run demo_nodes_cpp content_filtering_publisher
[INFO] [1715665128.560580531] [content_filtering_publisher]: Publishing: '-100.000000'
[INFO] [1715665129.560567925] [content_filtering_publisher]: Publishing: '-90.000000'
[INFO] [1715665130.560582011] [content_filtering_publisher]: Publishing: '-80.000000'
[INFO] [1715665131.560667908] [content_filtering_publisher]: Publishing: '-70.000000'
[INFO] [1715665132.560678495] [content_filtering_publisher]: Publishing: '-60.000000'
[INFO] [1715665133.560574546] [content_filtering_publisher]: Publishing: '-50.000000'
[INFO] [1715665134.560583367] [content_filtering_publisher]: Publishing: '-40.000000'
[INFO] [1715665135.560639324] [content_filtering_publisher]: Publishing: '-30.000000'
[INFO] [1715665136.560669182] [content_filtering_publisher]: Publishing: '-20.000000'
[INFO] [1715665137.560645125] [content_filtering_publisher]: Publishing: '-10.000000'
[INFO] [1715665138.560555876] [content_filtering_publisher]: Publishing: '0.000000'
[INFO] [1715665139.560552118] [content_filtering_publisher]: Publishing: '10.000000'
[INFO] [1715665140.560552912] [content_filtering_publisher]: Publishing: '20.000000'
[INFO] [1715665141.560622883] [content_filtering_publisher]: Publishing: '30.000000'
[INFO] [1715665142.560593886] [content_filtering_publisher]: Publishing: '40.000000'
[INFO] [1715665143.560606244] [content_filtering_publisher]: Publishing: '50.000000'
[INFO] [1715665144.560577487] [content_filtering_publisher]: Publishing: '60.000000'
[INFO] [1715665145.560645914] [content_filtering_publisher]: Publishing: '70.000000'
[INFO] [1715665146.560590296] [content_filtering_publisher]: Publishing: '80.000000'
[INFO] [1715665147.560582890] [content_filtering_publisher]: Publishing: '90.000000'
[INFO] [1715665148.560636555] [content_filtering_publisher]: Publishing: '100.000000'
[INFO] [1715665149.560581754] [content_filtering_publisher]: Publishing: '110.000000'
[INFO] [1715665150.560633471] [content_filtering_publisher]: Publishing: '120.000000'
[INFO] [1715665151.560667768] [content_filtering_publisher]: Publishing: '130.000000'
[INFO] [1715665152.560643344] [content_filtering_publisher]: Publishing: '140.000000'
[INFO] [1715665153.560564940] [content_filtering_publisher]: Publishing: '150.000000'
[INFO] [1715665154.560581222] [content_filtering_publisher]: Publishing: '-100.000000'
[INFO] [1715665155.560558818] [content_filtering_publisher]: Publishing: '-90.000000'
[INFO] [1715665156.560527694] [content_filtering_publisher]: Publishing: '-80.000000'
[INFO] [1715665157.560617218] [content_filtering_publisher]: Publishing: '-70.000000'
[INFO] [1715665158.560632347] [content_filtering_publisher]: Publishing: '-60.000000'
[INFO] [1715665159.560594403] [content_filtering_publisher]: Publishing: '-50.000000'
[INFO] [1715665160.560643862] [content_filtering_publisher]: Publishing: '-40.000000'
[INFO] [1715665161.560563158] [content_filtering_publisher]: Publishing: '-30.000000'
[INFO] [1715665162.560630742] [content_filtering_publisher]: Publishing: '-20.000000'
[INFO] [1715665163.560586125] [content_filtering_publisher]: Publishing: '-10.000000'
[INFO] [1715665164.560556959] [content_filtering_publisher]: Publishing: '0.000000'
[INFO] [1715665165.560578089] [content_filtering_publisher]: Publishing: '10.000000'
^C[INFO] [1715665166.407895487] [rclcpp]: signal_handler(signum=2)

root@tomoyafujita:~# ros2 run demo_nodes_cpp content_filtering_subscriber
[INFO 1715665136.743880612] [content_filtering_subscriber]: subscribed to topic "/temperature" with content filter options "data < %0 OR data > %1, {-30.000000, 100.000000}" (ContentFilteringSubscriber() at ./src/topics/content_filtering_subscriber.cpp:68)
[INFO 1715665149.561274129] [content_filtering_subscriber]: I receive an emergency temperature data: [110.000000] (operator()() at ./src/topics/content_filtering_subscriber.cpp:45)
[INFO 1715665150.561260423] [content_filtering_subscriber]: I receive an emergency temperature data: [120.000000] (operator()() at ./src/topics/content_filtering_subscriber.cpp:45)
[INFO 1715665151.561209019] [content_filtering_subscriber]: I receive an emergency temperature data: [130.000000] (operator()() at ./src/topics/content_filtering_subscriber.cpp:45)
[INFO 1715665152.561119757] [content_filtering_subscriber]: I receive an emergency temperature data: [140.000000] (operator()() at ./src/topics/content_filtering_subscriber.cpp:45)
[INFO 1715665153.561094590] [content_filtering_subscriber]: I receive an emergency temperature data: [150.000000] (operator()() at ./src/topics/content_filtering_subscriber.cpp:45)
[INFO 1715665154.561185095] [content_filtering_subscriber]: I receive an emergency temperature data: [-100.000000] (operator()() at ./src/topics/content_filtering_subscriber.cpp:45)
[INFO 1715665155.560752738] [content_filtering_subscriber]: I receive an emergency temperature data: [-90.000000] (operator()() at ./src/topics/content_filtering_subscriber.cpp:45)
[INFO 1715665156.560745138] [content_filtering_subscriber]: I receive an emergency temperature data: [-80.000000] (operator()() at ./src/topics/content_filtering_subscriber.cpp:45)
[INFO 1715665157.561193547] [content_filtering_subscriber]: I receive an emergency temperature data: [-70.000000] (operator()() at ./src/topics/content_filtering_subscriber.cpp:45)
[... cut ...]
[INFO 1715665159.561244053] [content_filtering_subscriber]: I receive an emergency temperature data: [-50.000000] (operator()() at ./src/topics/content_filtering_subscriber.cpp:45)
[INFO 1715665160.561297075] [content_filtering_subscriber]: I receive an emergency temperature data: [-40.000000] (operator()() at ./src/topics/content_filtering_subscriber.cpp:45)
^C[INFO 1715665165.791777362] [rclcpp]: signal_handler(signum=2) (signal_handler() at ./src/rclcpp/signal_handler.cpp:71)

```

</details>