osrf/ros2_test_cases #1137: Intra Process Communication

marcoag, 2024-04-29:
Check the documentation for the 'Intra Process Communication' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Intra Process Communication page](https://docs.ros.org/en/jazzy/Tutorials/Demos/Intra-Process-Communication.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-14:
@marcoag i can check this.

fujitatomoya, 2024-05-14:
no problem observed, we can close this.

<details>
  <summary>test result(click to expand!)</summary>

```shell
root@tomoyafujita:~# ros2 run intra_process_demo two_node_pipeline
Published message with value: 0, and address: 0x5DF2E859F190
 Received message with value: 0, and address: 0x5DF2E859F190
Published message with value: 1, and address: 0x5DF2E859F1B0
 Received message with value: 1, and address: 0x5DF2E859F1B0
Published message with value: 2, and address: 0x5DF2E859F190
 Received message with value: 2, and address: 0x5DF2E859F190
Published message with value: 3, and address: 0x5DF2E859F1B0
 Received message with value: 3, and address: 0x5DF2E859F1B0
Published message with value: 4, and address: 0x5DF2E859F190
 Received message with value: 4, and address: 0x5DF2E859F190
Published message with value: 5, and address: 0x5DF2E859F1B0
 Received message with value: 5, and address: 0x5DF2E859F1B0

root@tomoyafujita:~# ros2 run intra_process_demo cyclic_pipeline
Published first message with value:  42, and address: 0x63B1AE8A9930
Received message with value:         42, and address: 0x63B1AE8A9930
  sleeping for 1 second...
  done.
Incrementing and sending with value: 43, and address: 0x63B1AE8A9930
Received message with value:         43, and address: 0x63B1AE8A9930
  sleeping for 1 second...
  done.
Incrementing and sending with value: 44, and address: 0x63B1AE8A9930
Received message with value:         44, and address: 0x63B1AE8A9930
  sleeping for 1 second...
  done.
Incrementing and sending with value: 45, and address: 0x63B1AE8A9930
Received message with value:         45, and address: 0x63B1AE8A9930
  sleeping for 1 second...
  done.
Incrementing and sending with value: 46, and address: 0x63B1AE8A9930
Received message with value:         46, and address: 0x63B1AE8A9930
  sleeping for 1 second...
  done.
Incrementing and sending with value: 47, and address: 0x63B1AE8A9930
Received message with value:         47, and address: 0x63B1AE8A9930
  sleeping for 1 second...
```

</details>

[image]

[image]

closed 2024-05-14 by ahcorde