osrf/ros2_test_cases #1138: Logging and logger configuration

marcoag, 2024-04-29:
Check the documentation for the 'Logging and logger configuration' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Logging and logger configuration page](https://docs.ros.org/en/jazzy/Tutorials/Demos/Logging-and-logger-configuration.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-14:
@marcoag i can take care of this.

fujitatomoya, 2024-05-14:
Everything is fine except https://github.com/ros2/ros2_documentation/pull/4439

<details>
  <summary>test result(click to expand!)</summary>

```shell
root@tomoyafujita:~/jazzy_ws# ros2 run logging_demo logging_demo_main
[INFO] [1715663922.844733351] [logger_usage_demo]: Timer callback called (this will only log once)
[INFO] [1715663922.844973445] [logger_usage_demo]: Publishing: 'Current count: 0'
[INFO] [1715663923.344817642] [logger_usage_demo]: Publishing: 'Current count: 1'
[INFO] [1715663923.844715828] [logger_usage_demo]: Publishing: 'Current count: 2'
[INFO] [1715663924.344741748] [logger_usage_demo]: Publishing: 'Current count: 3'
[INFO] [1715663924.844735297] [logger_usage_demo]: Publishing: 'Current count: 4'
[INFO] [1715663925.344682563] [logger_usage_demo]: Publishing: 'Current count: 5'
[INFO] [1715663925.844781375] [logger_usage_demo]: Publishing: 'Current count: 6'
[INFO] [1715663926.344837343] [logger_usage_demo]: Publishing: 'Current count: 7'
[INFO] [1715663926.844716812] [logger_usage_demo]: Publishing: 'Current count: 8'
[INFO] [1715663927.344708802] [logger_usage_demo]: Publishing: 'Current count: 9'
[INFO] [1715663927.844821325] [logger_usage_demo]: Setting severity threshold to DEBUG
[INFO] [1715663927.844984222] [logger_usage_demo]: Publishing: 'Current count: 10'
[DEBUG] [1715663927.845026776] [logger_usage_demo]: Count is even (expression evaluated to true)
[INFO] [1715663928.344726524] [logger_usage_demo]: Publishing: 'Current count: 11'
[INFO] [1715663928.844791445] [logger_usage_demo]: Publishing: 'Current count: 12'
[DEBUG] [1715663928.844931611] [logger_usage_demo]: Count divides into 12 (function evaluated to true)
[DEBUG] [1715663928.844952930] [logger_usage_demo]: Count is even (expression evaluated to true)
[INFO] [1715663929.344831177] [logger_usage_demo]: Publishing: 'Current count: 13'
[INFO] [1715663929.844676125] [logger_usage_demo]: Publishing: 'Current count: 14'
[DEBUG] [1715663929.844730378] [logger_usage_demo]: Count is even (expression evaluated to true)
[INFO] [1715663930.344761403] [logger_usage_demo]: Publishing: 'Current count: 15'
[WARN] [1715663930.344890785] [logger_usage_demo]: Reseting count to 0
[INFO] [1715663930.844726009] [logger_usage_demo]: Publishing: 'Current count: 0'
[ERROR] [1715663930.844856451] [logger_usage_demo]: Modulo divisor cannot be 0
[DEBUG] [1715663930.844916816] [logger_usage_demo]: Count is even (expression evaluated to true)
[INFO] [1715663931.344824032] [logger_usage_demo]: Publishing: 'Current count: 1'
[DEBUG] [1715663931.344995505] [logger_usage_demo]: Count divides into 12 (function evaluated to true)

root@tomoyafujita:~# ros2 run logging_demo logging_demo_main
[INFO] [1715663959.145889866] [logger_usage_demo]: Timer callback called (this will only log once)
[INFO] [1715663959.146128966] [logger_usage_demo]: Publishing: 'Current count: 0'
[INFO] [1715663959.645898210] [logger_usage_demo]: Publishing: 'Current count: 1'
[INFO] [1715663960.145956710] [logger_usage_demo]: Publishing: 'Current count: 2'
[INFO] [1715663960.645958298] [logger_usage_demo]: Publishing: 'Current count: 3'
[INFO] [1715663961.145976376] [logger_usage_demo]: Publishing: 'Current count: 4'
[INFO] [1715663961.645949529] [logger_usage_demo]: Publishing: 'Current count: 5'
[INFO] [1715663962.145957447] [logger_usage_demo]: Publishing: 'Current count: 6'
[INFO] [1715663962.645952402] [logger_usage_demo]: Publishing: 'Current count: 7'
[INFO] [1715663963.145993854] [logger_usage_demo]: Publishing: 'Current count: 8'
[INFO] [1715663963.645905189] [logger_usage_demo]: Publishing: 'Current count: 9'
[INFO] [1715663964.145953895] [logger_usage_demo]: Setting severity threshold to DEBUG
[INFO] [1715663964.146106064] [logger_usage_demo]: Publishing: 'Current count: 10'
[DEBUG] [1715663964.146146125] [logger_usage_demo]: Count is even (expression evaluated to true)
[INFO] [1715663964.645952869] [logger_usage_demo]: Publishing: 'Current count: 11'
[INFO] [1715663965.145986826] [logger_usage_demo]: Publishing: 'Current count: 12'
[DEBUG] [1715663965.146124233] [logger_usage_demo]: Count divides into 12 (function evaluated to true)
[DEBUG] [1715663965.146145247] [logger_usage_demo]: Count is even (expression evaluated to true)
[INFO] [1715663965.645960671] [logger_usage_demo]: Publishing: 'Current count: 13'
[INFO] [1715663966.145953557] [logger_usage_demo]: Publishing: 'Current count: 14'
[DEBUG] [1715663966.146084228] [logger_usage_demo]: Count is even (expression evaluated to true)
[INFO] [1715663966.645957417] [logger_usage_demo]: Publishing: 'Current count: 15'
[WARN] [1715663966.646094919] [logger_usage_demo]: Reseting count to 0
[INFO] [1715663967.146008059] [logger_usage_demo]: Publishing: 'Current count: 0'
[ERROR] [1715663967.146145063] [logger_usage_demo]: Modulo divisor cannot be 0
[DEBUG] [1715663967.146222907] [logger_usage_demo]: Count is even (expression evaluated to true)
[INFO] [1715663967.645962590] [logger_usage_demo]: Publishing: 'Current count: 1'
[DEBUG] [1715663967.646109295] [logger_usage_demo]: Count divides into 12 (function evaluated to true)
[INFO] [1715663968.146006843] [logger_usage_demo]: Publishing: 'Current count: 2'
[DEBUG] [1715663968.146148105] [logger_usage_demo]: Count divides into 12 (function evaluated to true)
[DEBUG] [1715663968.146170083] [logger_usage_demo]: Count is even (expression evaluated to true)
[INFO] [1715663968.645955160] [logger_usage_demo]: Publishing: 'Current count: 3'
[... cut ...]
[INFO] [1715664551.785235062] [talker]: Publishing: 'Hello World: 1'
[INFO] [1715664552.785184214] [talker]: Publishing: 'Hello World: 2'
[INFO] [1715664553.785225364] [talker]: Publishing: 'Hello World: 3'
[INFO] [1715664554.785220949] [talker]: Publishing: 'Hello World: 4'
[INFO] [1715664555.785195518] [talker]: Publishing: 'Hello World: 5'
^C[INFO] [1715664556.170727553] [rclcpp]: signal_handler(signum=2)
root@tomoyafujita:~# ls .ros/log/talker_*
.ros/log/talker_2664_1715664550776.log
```

</details>

closed 2024-05-14 by ahcorde