osrf/ros2_test_cases #1139: Managed Nodes

marcoag, 2024-04-29:
Check the documentation for the 'Managed Nodes' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Managed Nodes page](https://docs.ros.org/en/jazzy/Tutorials/Demos/Managed-Nodes.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-14:
@marcoag i can check this, sign me up.

fujitatomoya, 2024-05-14:
No problem observed in the procedure, we can close this.

<details>
  <summary>test result(click to expand!)</summary>

```shell
root@tomoyafujita:~/jazzy_ws# ros2 launch lifecycle lifecycle_demo_launch.py
...
[INFO] [lifecycle_service_client-3]: process has finished cleanly [pid 2319]
[INFO] [launch]: process[lifecycle_service_client-3] was required: shutting down launched system
[INFO] [lifecycle_listener-2]: sending signal 'SIGINT' to process[lifecycle_listener-2]
[INFO] [lifecycle_talker-1]: sending signal 'SIGINT' to process[lifecycle_talker-1]
[lifecycle_listener-2] [INFO] [1715663636.070837355] [rclcpp]: signal_handler(signum=2)
[lifecycle_talker-1] [INFO] [1715663636.071198081] [rclcpp]: signal_handler(signum=2)
[lifecycle_talker-1] [ERROR] [1715663636.071573917] [lc_talker]: Unable to start transition 5 from current state shuttingdown: Could not publish transition: publisher's context is invalid, at ./src/rcl/publisher.c:423, at ./src/rcl_lifecycle.c:368
[lifecycle_talker-1] [WARN] [1715663636.071583909] [rclcpp_lifecycle]: Shutdown error in destruction of LifecycleNode: final state(unconfigured)
[INFO] [lifecycle_listener-2]: process has finished cleanly [pid 2318]
[INFO] [lifecycle_talker-1]: process has finished cleanly [pid 2317]

root@tomoyafujita:~# ros2 lifecycle get /lc_talker
active [3]
root@tomoyafujita:~# ros2 lifecycle set /lc_talker configure
Unknown transition requested, available ones are:
- deactivate [4]
- shutdown [7]
root@tomoyafujita:~# ros2 lifecycle list lc_talker
- cleanup [2]
	Start: inactive
	Goal: cleaningup
- activate [3]
	Start: inactive
	Goal: activating
- shutdown [6]
	Start: inactive
	Goal: shuttingdown
root@tomoyafujita:~# ros2 lifecycle list lc_talker -a
- configure [1]
	Start: unconfigured
	Goal: configuring
- transition_success [10]
	Start: configuring
	Goal: inactive
- transition_failure [11]
	Start: configuring
	Goal: unconfigured
- transition_error [12]
	Start: configuring
	Goal: errorprocessing
- cleanup [2]
	Start: inactive
	Goal: cleaningup
- transition_success [20]
	Start: cleaningup
	Goal: unconfigured
- transition_failure [21]
	Start: cleaningup
	Goal: inactive
- transition_error [22]
	Start: cleaningup
	Goal: errorprocessing
- activate [3]
	Start: inactive
	Goal: activating
- transition_success [30]
	Start: activating
	Goal: active
- transition_failure [31]
	Start: activating
	Goal: inactive
- transition_error [32]
[... 127 lines cut ...]
# Depending on which return value, the transition will be executed correctly or
# fallback/error callbacks will be triggered.

# The transition callback successfully performed its required functionality.
uint8 TRANSITION_CALLBACK_SUCCESS = 97

# The transition callback failed to perform its required functionality.
uint8 TRANSITION_CALLBACK_FAILURE = 98

# The transition callback encountered an error that requires special cleanup, if
# possible.
uint8 TRANSITION_CALLBACK_ERROR = 99

##
## Fields
##

# The transition id from above definitions.
uint8 id

# A text label of the transition.
string label
```

</details>