osrf/ros2_test_cases #1140: Quality of Service

marcoag, 2024-04-29:
Check the documentation for the 'Quality of Service' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Quality of Service page](https://docs.ros.org/en/jazzy/Tutorials/Demos/Quality-of-Service.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-14:
@marcoag i can take this.

fujitatomoya, 2024-05-14:
@marcoag @clalancette @ahcorde 

most of the procedure worked without any problems.

but i would like to fix a couple of minor things with https://github.com/ros2/ros2_documentation/pull/4438.

ahcorde, 2024-05-14:
Thank you @fujitatomoya

closed 2024-05-14 by ahcorde