osrf/ros2_test_cases #1143: dummy robot demo

marcoag, 2024-04-29:
Check the documentation for the 'dummy robot demo' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [dummy robot demo page](https://docs.ros.org/en/jazzy/Tutorials/Demos/dummy-robot-demo.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

Benjamin-Tan, 2024-05-04:
Could anyone assign this ticket to me?

Able to run this demo successfully.

### Step 1 - I was able to follow the documentation.
- It is unclear where this demo (dummy_robot_bringup) package comes from. A link to https://github.com/ros2/demos/tree/jazzy would be very much helpful. Should the demos be built from source or installed via apt?
- `mkdir -p ros2_ws/src`
- `cd ros2_ws/src`
- `git clone https://github.com/ros2/demos/tree/jazzy`
- `cd .. && colcon build --packages-up-to dummy_robot_bringup`

Step 2 - The documentation seemed clear to me.
- they are clear apart from the comment from above.

Step 3 - The documentation didn't have any obvious errors.
- the prints are outdated, the following is the updated prints.
    ```
    [INFO] [launch]: Default logging verbosity is set to INFO
    [INFO] [dummy_map_server-1]: process started with pid [2922]
    [INFO] [robot_state_publisher-2]: process started with pid [2923]
    [INFO] [dummy_joint_states-3]: process started with pid [2924]
    [INFO] [dummy_laser-4]: process started with pid [2925]
    [dummy_laser-4] [INFO] [1714837459.645517297] [dummy_laser]: angle inc:	0.004363
    [dummy_laser-4] [INFO] [1714837459.645613393] [dummy_laser]: scan size:	1081
    [dummy_laser-4] [INFO] [1714837459.645626640] [dummy_laser]: scan time increment: 	0.000000
    [robot_state_publisher-2] [INFO] [1714837459.652977937] [robot_state_publisher]: Robot initialized
    ```

ahcorde, 2024-05-06:
do you mind to open a PR updating the messages ?

Benjamin-Tan, 2024-05-06:
I don't mind, do I update to the rolling or jazzy branch?

clalancette, 2024-05-06:
> I don't mind, do I update to the rolling or jazzy branch?

Target the `rolling` branch to start with, then we'll backport to `jazzy` as necessary.  Thank you!

clalancette, 2024-05-06:
Thanks for the update.  Given that we have the open PR, I'm going to go ahead and close this one.

closed 2024-05-06 by clalancette