osrf/ros2_test_cases #1145: Composition

marcoag, 2024-04-29:
Check the documentation for the 'Composition' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Composition page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Composition.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

Benjamin-Tan, 2024-05-12:
1. I was able to follow the documentation
- Without the knowledge of where the composition package is from, it is not able to continue the tutorial.
- I could raise a PR to update it similarly to https://github.com/osrf/ros2_test_cases/issues/1143, but is there a reason why the debian package is not available for jazzy yet? I could only build it from source.

Apart from the above fixes, the entire composition demo is working as expected, without any obvious errors.

clalancette, 2024-05-12:
> I could raise a PR to update it similarly to https://github.com/osrf/ros2_test_cases/issues/1143, but is there a reason why > the debian package is not available for jazzy yet? I could only build it from source. The required debian packages are:
> 
>     composition
>     example_interfaces
>     image_tools

Can you explain more?  All of those packages are available for Jazzy:

```
$ sudo apt-cache policy ros-jazzy-composition
ros-jazzy-composition:
  Installed: 0.33.2-2noble.20240502.162905
  Candidate: 0.33.2-2noble.20240502.162905
  Version table:
 *** 0.33.2-2noble.20240502.162905 500
        500 http://packages.ros.org/ros2-testing/ubuntu noble/main amd64 Packages
        100 /var/lib/dpkg/status
$ sudo apt-cache policy ros-jazzy-example-interfaces
ros-jazzy-example-interfaces:
  Installed: 0.12.0-3noble.20240424.192404
  Candidate: 0.12.0-3noble.20240424.192404
  Version table:
 *** 0.12.0-3noble.20240424.192404 500
        500 http://packages.ros.org/ros2-testing/ubuntu noble/main amd64 Packages
        100 /var/lib/dpkg/status
$ sudo apt-cache policy ros-jazzy-image-tools
ros-jazzy-image-tools:
  Installed: 0.33.2-2noble.20240502.161104
  Candidate: 0.33.2-2noble.20240502.161104
  Version table:
 *** 0.33.2-2noble.20240502.161104 500
        500 http://packages.ros.org/ros2-testing/ubuntu noble/main amd64 Packages
        100 /var/lib/dpkg/status
```

Benjamin-Tan, 2024-05-12:
Opps I had some typos and hence not finding them, my bad. I'll raise a PR to add the installation instruction.

closed 2024-05-13 by clalancette