osrf/ros2_test_cases #1146: Creating an Action

marcoag, 2024-04-29:
Check the documentation for the 'Creating an Action' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Creating an Action page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Creating-an-Action.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

slowrunner, 2024-05-06:
For the config:
- Chip set: AMD64
- Build type: Debian
- OS: Ubuntu Noble 24.04
- DDS: FastDDS

I followed the "[Creating an Action](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Creating-an-Action.html)" tutorial:

1) Successfully followed the doc
2) Doc was clear:
- One thing confused me was in https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Creating-an-Action.html#building-an-action  -  
the phrase "before the ament_package() line, in the custom_action_interfaces:"  - before the ament_package() is very clear but I wasn't sure what "in the custom_action_interfaces:"  meant.  If it had said "in the custom_action_interfaces package.xml file", I would not have expected I need to find "custom_action_interfaces:" in the package.xml file.
- Another thing confused me - why do here I source the install/local_setup.bash instead of the usual install/setup.bash?

3) The doc had no errors

```
$ source install/local_setup.bash
$ ros2 interface show custom_action_interfaces/action/Fibonacci 

```

marcoag, 2024-05-07:
I believe the idea of using `local_setup.bash` is to make sure it only includes the packages in the `install` space. You can find more info on it on the [ament documentation](https://design.ros2.org/articles/ament.html) on the section of environment creation.

Will close this as completed, thanks  a lot for testing!

closed 2024-05-07 by marcoag