osrf/ros2_test_cases #1147: Creating Launch Files

marcoag, 2024-04-29:
Check the documentation for the 'Creating Launch Files' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Creating Launch Files page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Launch/Creating-Launch-Files.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

knmcguire, 2024-05-07:
_Step 1 - I was able to follow the documentation._ -> Yes
_Step 2 - The documentation seemed clear to me._ -> Yes
_Step 3 - The documentation didn't have any obvious errors._ ->Seems to be all good! Sometimes I forgot to source my environment again but that is on me.

**Seems all good**

_Some thoughts for improvement_

When examining the launch file, it is more useful to first go through it code for code and then explain the global behavior. 

Also the rqt graph wan't the same as the picture (mine didn't have the /launch_ros

[image]

clalancette, 2024-05-07:
> When examining the launch file, it is more useful to first go through it code for code and then explain the global behavior.

I'm a bit confused by this comment.  Isn't this what the current tutorial does?  Could you explain a bit more what you mean here, or even better open a PR with changes?

knmcguire, 2024-05-07:
Ah if you look at that current section, it first let's the user first copy the full code, then there is a global explanation and then there is the line by line explanation. To switch that around (switch line by line with global explanation) would perhaps improve the flow better. But inherently the tutorial itself still holds according to the test. 

I can propose an PR for the suggestion to the documentation itself. You could close this issue.

clalancette, 2024-05-07:
> I can propose an PR for the suggestion to the documentation itself. You could close this issue.

Sounds good, thank you!

closed 2024-05-07 by clalancette