osrf/ros2_test_cases #1149: Launch system

marcoag, 2024-04-29:
Check the documentation for the 'Launch system' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Launch system page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Launch/Launch-system.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

knmcguire, 2024-05-10:
I'll be trying this out on a Jazzy docker. You can assign me

knmcguire, 2024-05-10:
I build the launch.py with a python package and the cpp package with launch.xml and launch.yaml.

```
root@2ee5726f5fb2:/launch_ws# ros2 launch py_launch_example my_script_launch.py 
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-05-10-11-44-55-971091-2ee5726f5fb2-9632
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [talker-1]: process started with pid [9635]
[talker-1] [INFO] [1715341497.029241879] [talker]: Publishing: 'Hello World: 1'
[talker-1] [INFO] [1715341498.029205852] [talker]: Publishing: 'Hello World: 2'
[talker-1] [INFO] [1715341499.029214899] [talker]: Publishing: 'Hello World: 3'
```
```
root@2ee5726f5fb2:/launch_ws# ros2 launch cpp_launch_example my_script_launch.yaml 
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-05-10-11-44-02-318120-2ee5726f5fb2-8892
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [talker-1]: process started with pid [8898]
[talker-1] [INFO] [1715341443.423096350] [talker]: Publishing: 'Hello World: 1'
[talker-1] [INFO] [1715341444.422896026] [talker]: Publishing: 'Hello World: 2'
[talker-1] [INFO] [1715341445.422898285] [talker]: Publishing: 'Hello World: 3'
```
```
root@2ee5726f5fb2:/launch_ws# ros2 launch cpp_launch_example my_script_launch.xml 
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-05-10-11-43-06-490934-2ee5726f5fb2-8216
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [talker-1]: process started with pid [8219]
[talker-1] [INFO] [1715341387.597001688] [talker]: Publishing: 'Hello World: 1'
[talker-1] [INFO] [1715341388.597297622] [talker]: Publishing: 'Hello World: 2'
[talker-1] [INFO] [1715341389.596992190] [talker]: Publishing: 'Hello World: 3'
```

- [ ] Step 1 - I was able to follow the documentation.
- [ ] Step 2 - The documentation seemed clear to me.
- [ ] Step 3 - The documentation didn't have any obvious errors.

Seems to work ✅ 

_Very tiny nitpick._ My OCD senses tingled when I saw that the workspace was called `launch_ws `instead of `ros2_ws `as in the other tutorials but on the other hand it is good practice to separate workspaces based on projects anyway.

marcoag, 2024-05-10:
I believe we can leave with a different workspace name for this tutorial, as you said it's also good to keep them separated. 

Thanks for testing!

closed 2024-05-10 by marcoag