osrf/ros2_test_cases #1150: Using Event Handlers

marcoag, 2024-04-29:
Check the documentation for the 'Using Event Handlers' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Using Event Handlers page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Launch/Using-Event-Handlers.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

knmcguire, 2024-05-10:
You can assign me to this since I'm working on the launch tutorial series anyway.

knmcguire, 2024-05-10:
Done on a Jazzy docker image

```
# ros2 launch launch_tutorial example_event_handlers_launch.py turtlesim_ns:='turtlesim3' use_provided_red:='True' new_background_r:=200
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-05-10-14-32-50-017217-2ee5726f5fb2-32814
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [turtlesim_node-1]: process started with pid [32827]
[INFO] [launch.user]: Turtlesim started, spawning turtle
[INFO] [Spawn "{x: 2, y: 2, theta: 0.2}"-2]: process started with pid [32828]
[turtlesim_node-1] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[turtlesim_node-1] [INFO] [1715351570.155214234] [turtlesim3.sim]: Starting turtlesim with node name /turtlesim3/sim
[turtlesim_node-1] [INFO] [1715351570.163203381] [turtlesim3.sim]: Spawning turtle [turtle1] at x=[5.544445], y=[5.544445], theta=[0.000000]
[turtlesim_node-1] MESA: error: ZINK: vkCreateInstance failed (VK_ERROR_INCOMPATIBLE_DRIVER)
[turtlesim_node-1] glx: failed to create drisw screen
[INFO] [launch.user]: Spawn request says "requester: making request: turtlesim.srv.Spawn_Request(x=2.0, y=2.0, theta=0.2, name='')"
[turtlesim_node-1] [INFO] [1715351570.419792913] [turtlesim3.sim]: Spawning turtle [turtle2] at x=[2.000000], y=[2.000000], theta=[0.200000]
[INFO] [launch.user]: Spawn request says "response:
turtlesim.srv.Spawn_Response(name='turtle2')"
[INFO] [Spawn "{x: 2, y: 2, theta: 0.2}"-2]: process has finished cleanly [pid 32828]
[INFO] [launch.user]: Spawn finished
[INFO] [sim background_r 120-3]: process started with pid [32860]
[INFO] [sim background_r 120-3]: process has finished cleanly [pid 32860]
[INFO] [sim background_r 200-4]: process started with pid [32942]
[INFO] [sim background_r 200-4]: process has finished cleanly [pid 32942]
[INFO] [turtlesim_node-1]: process has finished cleanly [pid 32827]
```

- [ ] Step 1 - I was able to follow the documentation.
- [ ] Step 2 - The documentation seemed clear to me.
- [ ] Step 3 - The documentation didn't have any obvious errors.

**Seems good to me** ✅ 

Some additional thoughts and improvements:

I did think that this tutorial was easier to follow, even though there was more 'code' to be copied. It explained mostly about one concept, namely the different events of a lifecycle of a node. I would need to look closer on why the substitution tutorial (https://github.com/osrf/ros2_test_cases/issues/1152) was much harder to follow than this one.

clalancette, 2024-05-10:
Thanks for testing!

closed 2024-05-10 by clalancette