osrf/ros2_test_cases #1151: Using ROS2 Launch For Large Projects

marcoag, 2024-04-29:
Check the documentation for the 'Using ROS2 Launch For Large Projects' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Using ROS2 Launch For Large Projects page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Launch/Using-ROS2-Launch-For-Large-Projects.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

knmcguire, 2024-05-10:
You can assign me to this, since I've already started with this tutorial: https://github.com/osrf/ros2_test_cases/issues/1152

knmcguire, 2024-05-10:
Ah woops, I actually meant an different issue, but no worries this one is up next.

knmcguire, 2024-05-10:
- [ ] Step 1 - I was able to follow the documentation. 
- [ ] Step 2 - The documentation seemed clear to me.
- [ ] Step 3 - The documentation didn't have any obvious errors.

The tutorials is really long and I lost oversight at one point. Also there were errors so I couldn't complete the tutorial

**This needs fixing** 

I have here a list of suggestion for either myself or someone else to look at if they'd like to fix this:
* Section 2.2 If I follow through with the 'If we now start the turtlesim_world_2_launch.py launch file', I get the following error. This is because extra step of adding the config folder at the very end of the tutorial
`[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: /home/ubuntu/ros2_ws/install/launch_tutorial/share/launch_tutorial/config/turtlesim.yaml`

* Section 3: `turtlsesim_world_2 `and `turtle_world2_with_namespace `still needs to be added to `LaunchDescription([` as well.

* Running launch files
  * 1- Update setup.py. Forgot to add the Rviz folder.
  * 2- Build and Run: I get the following error. 
`[ERROR] [launch]: Caught exception in launch (see debug for traceback): environment variable 'USER' does not exist`
if I disabled that fixed_broadcaster node, I get the following warning

`[turtle_tf2_listener-6] [INFO] [1715366094.127739162] [listener]: Could not transform turtle2 to carrot1: "carrot1" passed to lookupTransform argument source_frame does not exist. `

Some global thoughts on why the flow of the tutorial wasn't great either. It might be too much in one tutorial to all show in one project/launch file. And if people come to this point, this might be too 'advanced' to be in the form of a tutorial. I'm wondering if it is better suited as an sort of API documentation or user-guide with minimal examples of each case., that can be referred to after users have gone through the basics (such they can do it later when they make their own project).  Might reduce the maintainer load of this tutorial/doc as well.

knmcguire, 2024-05-14:
There is a partial fix proposed in https://github.com/ros2/ros2_documentation/pull/4454, however, there is something wrong with the carrot1 transform... it seems to be missing in the full tutorial so I think we are missing a node here. 

Perhaps with this proposal and draft PR open, you could consider closing this issue.

knmcguire, 2024-05-17:
Ah ofcourse, the 'fixed_frame_node' also included the carrot1 broadcaster. User error! my bad

That covers of what I thought was the biggest error of the tutorial. That means that this issue can be closed with the PR https://github.com/ros2/ros2_documentation/pull/4454 just adding some small fixes.

clalancette, 2024-05-17:
> That covers of what I thought was the biggest error of the tutorial. That means that this issue can be closed with the PR [ros2/ros2_documentation#4454](https://github.com/ros2/ros2_documentation/pull/4454) just adding some small fixes.

Fantastic, thank you for opening that and for testing!