osrf/ros2_test_cases #1152: Using Substitutions

marcoag, 2024-04-29:
Check the documentation for the 'Using Substitutions' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Using Substitutions page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Launch/Using-Substitutions.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

knmcguire, 2024-05-10:
You can assign me to this. I'll be working on it with the provided Jazzy docker image

knmcguire, 2024-05-10:
- [ ] Step 1 - I was able to follow the documentation.
- [ ] Step 2 - The documentation seemed clear to me.
- [ ] Step 3 - The documentation didn't have any obvious errors.

knmcguire, 2024-05-10:
- [ ] Step 1 - I was able to follow the documentation. 
- [ ] Step 2 - The documentation seemed clear to me.
- [ ] Step 3 - The documentation didn't have any obvious errors.

I was able to complete the tutorial, but I found this tutorial extremely difficult to follow and I had to re-read a couple of paragraph multiple times... 

Currently I've only done the launch.py variant and the yaml seems to be a bit shorter but I think the main problem is that multiple concepts, which are already advanced on their own, are being explained all in one go. I remember as well when I had to figure this concept out a little while ago, It was quite difficult for me to wrap my head around what is actually happening. 

I'd like to finish up the rest of the launch tutorials first for the tutorial party, but for this one I'd like to look into rewriting this tutorial. This is quite an important feature of the launch functionality I think

knmcguire, 2024-05-14:
There are no errors in this tutorial of what I could find, other than that the tutorial is hard to follow. I tried to do a rewrite but I think  in general that the full launch series would need that and that is a much bigger endevor (see this issue: https://github.com/ros2/ros2_documentation/issues/4453), but I see that as out of scope of this tutorial party.

For now, this tutorial does not contain errors so In my opinion you can close this issue.

marcoag, 2024-05-16:
Yup, sounds good. Thanks for the testing and the issue with all the info!

closed 2024-05-16 by marcoag