osrf/ros2_test_cases #1153: Monitoring For Parameter Changes CPP

marcoag, 2024-04-29:
Check the documentation for the 'Monitoring For Parameter Changes CPP' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Monitoring For Parameter Changes CPP page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Monitoring-For-Parameter-Changes-CPP.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-09:
@marcoag i can take care of this.

fujitatomoya, 2024-05-09:
test result,

```bash
root@tomoyafujita:~/jazzy_ws/src/cpp_parameter_event_handler/src# ros2 run demo_nodes_cpp parameter_blackboard
[INFO] [1715236604.329082601] [parameter_blackboard]: Parameter blackboard node named '/parameter_blackboard' ready, and serving '6' parameters already!
^C[INFO] [1715236613.152169653] [rclcpp]: signal_handler(signum=2)

root@tomoyafujita:~/jazzy_ws# ros2 run cpp_parameter_event_handler parameter_event_handler
[INFO] [1715236609.744423569] [node_with_parameters]: cb2: Received an update to parameter "a_double_param" of type: double: "3.45"
^C[INFO] [1715236613.735915929] [rclcpp]: signal_handler(signum=2)

root@tomoyafujita:~# ros2 param set parameter_blackboard a_double_param 3.45
Set parameter successful
```

no error observed, we can close this.

closed 2024-05-09 by ahcorde