osrf/ros2_test_cases #1154: Monitoring For Parameter Changes Python

marcoag, 2024-04-29:
Check the documentation for the 'Monitoring For Parameter Changes Python' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Monitoring For Parameter Changes Python page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Monitoring-For-Parameter-Changes-Python.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-09:
@marcoag i can test this.

fujitatomoya, 2024-05-09:
test result,

```bash
root@tomoyafujita:~/jazzy_ws# ros2 run python_parameter_event_handler node_with_parameters
[INFO] [1715236143.070046736] [node_with_parameters]: Received an update to parameter: a_double_param: 3.45

root@tomoyafujita:~/jazzy_ws# ros2 run demo_nodes_cpp parameter_blackboard
[INFO] [1715236136.871653589] [parameter_blackboard]: Parameter blackboard node named '/parameter_blackboard' ready, and serving '6' parameters already!

root@tomoyafujita:~# ros2 param set parameter_blackboard a_double_param 3.45
Set parameter successful

```

no problem observed, we can close this issue.

closed 2024-05-09 by ahcorde