osrf/ros2_test_cases #1156: RViz Main

marcoag, 2024-04-29:
Check the documentation for the 'RViz Main' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [RViz Main page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/RViz/RViz-Main.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

jasmeet0915, 2024-05-08:
This is an index page. Both the links work correctly on the page!

clalancette, 2024-05-08:
Fantastic, thanks for testing!

closed 2024-05-08 by clalancette