osrf/ros2_test_cases #1157: RViz User Guide

marcoag, 2024-04-29:
Check the documentation for the 'RViz User Guide' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [RViz User Guide page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/RViz/RViz-User-Guide/RViz-User-Guide.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-09:
@marcoag i can check this.

fujitatomoya, 2024-05-09:
test result,

[image]

everything works okay including shortcut keys.

closed 2024-05-09 by ahcorde