osrf/ros2_test_cases #1158: Rosdep

marcoag, 2024-04-29:
Check the documentation for the 'Rosdep' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Rosdep page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Rosdep.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

knmcguire, 2024-05-07:
You can assign me to this

knmcguire, 2024-05-07:
**Step 1 - I was able to follow the documentation.**4
Yes I was

**Step 2 - The documentation seemed clear to me.**
Pretty clear to me!

**Step 3 - The documentation didn't have any obvious errors.**
Not that I could see. 

Also all code didn't contain errors.

**Seems all okay!**

_Some additional thoughts_ This doc seems to be more of an 'concepts' documentation than a tutorial, especially since it doesn't prove an example package to test on so it is mostly just explaining the concepts.

closed 2024-05-07 by ahcorde