osrf/ros2_test_cases #1159: CLI

marcoag, 2024-04-29:
Check the documentation for the 'CLI' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [CLI page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Testing/CLI.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

jasmeet0915, 2024-05-08:
I have tested all the commands mentioned in the tutorial with a demo package and they work as expected! All the links seem to be working as well

[image]

clalancette, 2024-05-08:
Thank you for testing!

closed 2024-05-08 by clalancette