osrf/ros2_test_cases #1160: Cpp

marcoag, 2024-04-29:
Check the documentation for the 'Cpp' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Cpp page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Testing/Cpp.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-09:
@marcoag i can take care of this.

fujitatomoya, 2024-05-09:
@marcoag https://github.com/ros2/ros2_documentation/pull/4412 can close this issue.

closed 2024-05-09 by ahcorde