osrf/ros2_test_cases #1161: Python

marcoag, 2024-04-29:
Check the documentation for the 'Python' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Python page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Testing/Python.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

jasmeet0915, 2024-05-08:
I have tested all the instructions in the tutorial with a demo ament_python package and everything works as expected! I just noted a minor typo for which I have opened up a [PR](https://github.com/ros2/ros2_documentation/pull/4408)

[image]

clalancette, 2024-05-08:
Thanks for testing!

closed 2024-05-08 by clalancette