osrf/ros2_test_cases #1163: Adding A Frame Cpp

marcoag, 2024-04-29:
Check the documentation for the 'Adding A Frame Cpp' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Adding A Frame Cpp page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Tf2/Adding-A-Frame-Cpp.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

knmcguire, 2024-05-17:
I was just testing this now on the jazzy docker.

knmcguire, 2024-05-17:
I'm able to follow the tutorial in full but It seems that something weird at the end of the the first part of the tutorial (and probably also at the end as well)

This is with `ros2 launch learning_tf2_cpp turtle_tf2_fixed_frame_demo_launch.py`

https://github.com/osrf/ros2_test_cases/assets/11061260/b27cddcd-5bdd-41c3-add9-7d590eac36e4

But with `ros2 launch learning_tf2_cpp turtle_tf2_fixed_frame_demo_launch.py target_frame:=carrot1` I get this behavior

https://github.com/osrf/ros2_test_cases/assets/11061260/832e514a-d7d7-4173-b846-b5a9823924cf

This is what it looks like in rviz2: 
[image]

Sometimes when I startup the example, it does sort of follow carrot1, and sometimes it is weirdly in between. I observed the same thing with the python example of this as well. I don't know if this an issue with the transform lookup or the controller...

knmcguire, 2024-05-17:
I've also tried the the packaged version of the fixed frame broadcaster by replacing this node:
```
Node(
    package='learning_tf2_cpp',
    executable='fixed_frame_tf2_broadcaster',
    name='fixed_broadcaster',
),

```
with
```
      Node(
            package='turtle_tf2_py',
            executable='fixed_frame_tf2_broadcaster',
            name=[LaunchConfiguration('node_prefix'), 'fixed_broadcaster'],
      ),
```

Then it looks more consistently like this. Still rotating but at least around carrot1 all the time

https://github.com/osrf/ros2_test_cases/assets/11061260/df8d1764-2682-4c2a-b46c-c64e1f5eadc3

knmcguire, 2024-05-17:
Hmm.. a restart seems to have 'solved' the issue. Now it looks more similar to the regular turtle_tf2_py package
Fixed frame:

Dynamic frame:

[image]

clalancette, 2024-05-17:
> Hmm.. a restart seems to have 'solved' the issue.

That sort of feels like there was another node still running in the background, also publishing to tf2, which was messing up the frames.  Hard to tell now, but that is my guess.

> Also the tutorial was good to follow. Seems all okay to me ✅

This is great, then.  Thanks for testing!

closed 2024-05-17 by clalancette

knmcguire, 2024-05-17:
yes that is true, but then I would have expected the regular turtle_tf2_py to also behave weirdly. Also when I looked at RVIZ2 at the time it was very stationary. But can't recreate it anyway