osrf/ros2_test_cases #1166: Introduction To Tf2

marcoag, 2024-04-29:
Check the documentation for the 'Introduction To Tf2' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Introduction To Tf2 page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Tf2/Introduction-To-Tf2.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

ottojo, 2024-05-01:
Please assign me this issue, i will go through the tutorial

ottojo, 2024-05-01:
No issues with the TF2 introduction using turtlesim and rviz!

## Environment
* `ros:jazzy` docker image
* linux `6.8.8-1-MANJARO`, `x86_64`

## Step 1 - I was able to follow the documentation.

### Installing the demo
* Packages all installed successfully using apt

### Running the demo
* Turtlesim launches without errors
* turtle_teleop_key starts and works

### tf2 tools
#### 1 Using view_frames
* view_frames creates pdf diagram of tf tree
* slight differences in terminal output and diagram layout, but not critical (logger prefix, node layout)

#### 2 Using tf2_echo
* tf2_echo works, output exactly matches tutorial

### rviz2 and tf2
* provided command starts rviz with correct configuration
* viewing exact numeric values of transforms requires unfolding items in rviz sidebar, looks exactly like in screenshot then

## Step 2 - The documentation seemed clear to me.
* all steps are clearly laid out, tf functionality is demonstrated

## Step 3 - The documentation didn't have any obvious errors.
* no errors found

## Notes
I was about to suggest introducing rqt-tf-tree in this tutorial, but that one actually crashed when i tried to use it just now :smile: I will see if an issue exists for that already...

clalancette, 2024-05-01:
> I was about to suggest introducing rqt-tf-tree in this tutorial, but that one actually crashed when i tried to use it just now 😄 I will see if an issue exists for that already...

`rqt_tf_tree` isn't one of the core tools; we typically stick with core tools in the tutorials.

In any case, thanks for the testing!  Closing this one out.

closed 2024-05-01 by clalancette

mikaelarguedas, 2024-05-05:
When testing this, the fact that tutorial and console output of `view_frames` referred to file that doesn't exist (frames.pdf) was a bit surprising https://github.com/ros2/geometry2/pull/674