osrf/ros2_test_cases #1167: Learning About Tf2 And Time Cpp

marcoag, 2024-04-29:
Check the documentation for the 'Learning About Tf2 And Time Cpp' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Learning About Tf2 And Time Cpp page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Tf2/Learning-About-Tf2-And-Time-Cpp.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

knmcguire, 2024-05-17:
currently following this tutorial so you can assign it to me

knmcguire, 2024-05-17:
Step 1 2 3 were all good.

Final result looked like this. I was also able to notice the delay when I increased the time from 50ms to 2s 

[image]

clalancette, 2024-05-17:
Thanks for testing!

closed 2024-05-17 by clalancette