osrf/ros2_test_cases #1171: Time Travel With Tf2 Cpp

marcoag, 2024-04-29:
Check the documentation for the 'Time Travel With Tf2 Cpp' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Time Travel With Tf2 Cpp page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Tf2/Time-Travel-With-Tf2-Cpp.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

knmcguire, 2024-05-17:
You can assign me to this as I'm now following this tutorial

knmcguire, 2024-05-17:
Step 1,2,3 all seems OK ✅ 

Result of first part of tutorial
[image]

Result of second part of tutorial
[image]
```

Just some thought for improvement, I had to read 3 times this part of the tutorial and understand the implications:

In our code we asked tf2 the following question: “What was the pose of carrot1 5 seconds ago, relative to turtle2 5 seconds ago?”. This means we are controlling the second turtle based on where it was 5 seconds ago as well as where the first carrot was 5 seconds ago.

However, what we really want to ask is: “What was the pose of carrot1 5 seconds ago, relative to the current position of the turtle2?”.
```

Perhaps something visual could have helped here. But its nitpicking mostly. you can close this issue.

marcoag, 2024-05-17:
Sounds great, thanks a lot for the testing! closing.

PS: Feel free to open a PR with your suggestion.

closed 2024-05-17 by marcoag