osrf/ros2_test_cases #1172: Time Travel With Tf2 Py

marcoag, 2024-04-29:
Check the documentation for the 'Time Travel With Tf2 Py' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Time Travel With Tf2 Py page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Tf2/Time-Travel-With-Tf2-Py.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-09:
@marcoag referenced URL does not exist, https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Tf2/Time-Travel-With-Tf2-Py.html

ahcorde, 2024-05-09:
we removed some tutorials https://github.com/ros2/ros2_documentation/pull/4384 because they never worked

closed 2024-05-09 by ahcorde