osrf/ros2_test_cases #1174: Writing A Tf2 Broadcaster Cpp

marcoag, 2024-04-29:
Check the documentation for the 'Writing A Tf2 Broadcaster Cpp' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Writing A Tf2 Broadcaster Cpp page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Broadcaster-Cpp.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

ottojo, 2024-05-02:
i'm testing and reviewing this tutorial, please assign me!

ottojo, 2024-05-02:
## Setup
`ros:jazzy` docker image

```
   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-6.8.8-1-MANJARO-x86_64-with-glibc2.39
release          : 6.8.8-1-MANJARO
processor        : x86_64

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : jazzy
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}
```

## Notes

* `turtle_tf2_broadcaster.cpp` is available and builds
* provided launchfile works
* explanation for node and launchfile is clear
* running turtlesim works, resulting TF output is as expected

clalancette, 2024-05-02:
Thank you for testing!

closed 2024-05-02 by clalancette