osrf/ros2_test_cases #1176: Writing A Tf2 Listener Cpp

marcoag, 2024-04-29:
Check the documentation for the 'Writing A Tf2 Listener Cpp' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Writing A Tf2 Listener Cpp page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Listener-Cpp.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

ottojo, 2024-05-01:
i can test this, please assign this to me :+1:

ottojo, 2024-05-13:
i am now finally getting to this, please do assign me

ottojo, 2024-05-13:
The provided code and commandline snippets all work as expected.

I opened a MR for a few notes i had during testing: https://github.com/ros2/ros2_documentation/pull/4434
* the part of actually using the TF goes completely unmentioned, i added a sentence at least to state the purpose of that bit of the code
* two minor syntax fixes, one too many `{}` inline and a whitespace change in the code
* code uses a `target_frame` parameter, which is later called `fromFrameRel` and then passed to the `source_frame` argument of `lookupTransform()`, which may be confusing. I really like the introduction clarifying relationships between frames versus transformations of points in frames in this docs pr https://github.com/ros2/ros2_documentation/pull/4235/, so we might link to that in the future

clalancette, 2024-05-13:
Thanks for testing!  Given that you have the open PR, I'm going to close this out as "done", and we can iterate on the PR.

closed 2024-05-13 by clalancette