osrf/ros2_test_cases #1177: Writing A Tf2 Listener Py

marcoag, 2024-04-29:
Check the documentation for the 'Writing A Tf2 Listener Py' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Writing A Tf2 Listener Py page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Listener-Py.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

mikaelarguedas, 2024-05-05:
Tutorial clear and completed successfully.

minor fixups at https://github.com/ros2/ros2_documentation/pull/4360

marcoag, 2024-05-06:
Closing and following up on the issue. thanks!

closed 2024-05-06 by marcoag