osrf/ros2_test_cases #1178: Writing A Tf2 Static Broadcaster Cpp

marcoag, 2024-04-29:
Check the documentation for the 'Writing A Tf2 Static Broadcaster Cpp' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Writing A Tf2 Static Broadcaster Cpp page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Static-Broadcaster-Cpp.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

ottojo, 2024-05-01:
i'm going through this tutorial, please assign this to me

ottojo, 2024-05-01:
The tutorial works as intended on jazzy and is straightforward to follow :+1: 

## Environment
* `ros:jazzy` docker image
* linux `6.8.8-1-MANJARO`, `x86_64`

## Notes

### Tasks

#### 1 Create a package
* provided command works, creates package as explained

#### 3 Build
* rosdep wasn't initialized, but shows instructions on how to do that, maybe we have a rosdep tutorial we can link to?
* building worked as described
* sourcing the workspace uses `. install/setup.bash` syntax instead of `source`, which works but is perhaps confusing and not in line with other tutorials. (edit: nevermind, i found 49 occurrences of `. install/...` and 50 occurrences of `source install/` in the docs sources, so i guess there isnt a *right* way :laughing: )

#### 4 Run
* Running the node works
* static tf appears when echoing /tf_static

### The proper way to publish static transforms
 * commands for `static_transform_publisher` work
 * launchfile example works without modification

clalancette, 2024-05-01:
Thanks for testing!

closed 2024-05-01 by clalancette