osrf/ros2_test_cases #1179: Writing A Tf2 Static Broadcaster Py

marcoag, 2024-04-29:
Check the documentation for the 'Writing A Tf2 Static Broadcaster Py' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Writing A Tf2 Static Broadcaster Py page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Static-Broadcaster-Py.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

mikaelarguedas, 2024-05-05:
Tutorial clear and completed successfully.

minor fixups at https://github.com/ros2/ros2_documentation/pull/4360

mikaelarguedas, 2024-05-05:
Question @clalancette :

The tutorial mentions license:
```
<license>Apache License 2.0</license>
```

The license identifier generated by `ros2 pkg create` is the spdx version : 
```
<license>Apache-2.0</license>
```

Which format should be used ? Is there a prefered one ?

clalancette, 2024-05-07:
> Which format should be used ? Is there a prefered one ?

We slightly prefer the SPDX identifier (Apache-2.0), though we are not consistent about this, even in the core.

clalancette, 2024-05-07:
I'm going to go ahead and close this, as the question about the license is somewhat larger (which is hinted at by https://github.com/ros-infrastructure/rep/pull/347 as well).  Thanks for testing!

closed 2024-05-07 by clalancette

mikaelarguedas, 2024-05-08:
Alright thanks,

PR for license mismatch in tutorials here https://github.com/ros2/ros2_documentation/pull/4395