osrf/ros2_test_cases #1183: Exporting an URDF File

marcoag, 2024-04-29:
Check the documentation for the 'Exporting an URDF File' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Exporting an URDF File page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/URDF/Exporting-an-URDF-File.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

jasmeet0915, 2024-05-07:
I have tested all the links in this page and 2 of them seem to be incorrect:

- sdformat_urdf page points to a non-existant page. I have corrected the link and opened up a [PR](https://github.com/ros2/ros2_documentation/pull/4394).
- The link to the URDF Webviewer again points to the [ExportURDF](https://github.com/daviddorf2023/ExportURDF) GitHub page which I think is incorrect. Which web viewer was this intended for?

marcoag, 2024-05-08:
Good catch. For the first one please add the fix to `rolling`, [as suggested in the comment](https://github.com/ros2/ros2_documentation/pull/4394#issuecomment-2099417087). For the second one I see it was added [here](https://github.com/ros2/ros2_documentation/pull/4273/files), I found https://github.com/gkjohnson/urdf-loaders/ as an option but maybe @kscottz can tell us the exact one...

kscottz, 2024-05-08:
[I think this is what I planning to point to.](https://gkjohnson.github.io/urdf-loaders/javascript/example/bundle/index.html)

clalancette, 2024-05-08:
We've now merged in all of the fixes for this, so I'm going to go ahead and close it.  Thanks for the fixes!

closed 2024-05-08 by clalancette