osrf/ros2_test_cases #1185: Using URDF with Robot State Publisher

marcoag, 2024-04-29:
Check the documentation for the 'Using URDF with Robot State Publisher' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Using URDF with Robot State Publisher page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/URDF/Using-URDF-with-Robot-State-Publisher.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

mdxtinkernick, 2024-05-01:
I can do this one

mdxtinkernick, 2024-05-01:
I was able to follow the tutorial, but with some error, and some questions about style and consistency below 

The issues started in the final steps 

6 Install the package

cd second_ros2_ws

didn't work as the terminal is not in the home folder at this point - in step 1 you moved the terminal into the package - needs a full path or initial cd by itself - not sure if there is a preferred approach to that

and then in

7 view the results

it is again assuming the terminal is in the home folder - but the last command was run inside the workspace, opening a new tab for me opens in the same location the tab you were in was so it does not find the rviz config file, which fails silently and rviz pops up with nothing visualised.

At this point it was a bit confusing as I looked at the end of the path and saw it should be in a folder called urdf_tutorial_r2d2 - but we were told before to save it in the urdf folder. 

So relative paths problematic, slightly confusing moving around of file in the copying stage of the build.

I also found the switching between copying and pasting commands, and at times following written instructions to issue commands inconsistent.

Also downloading some files, and copying and pasting the content for others meant I had to think more about the mechanics of what was being done.

Not errors -

section 5
don't need to import setup and find_packages as they are already there

deprecation message about easy install on colcon build - not sure if that is specific to this package or building in general

marcoag, 2024-06-26:
@mdxtinkernick thanks for the time to test and the report! personally I think it's good that people are aware with the mechanics of what's being done instead of brainlessly copy/pasting commands. With that said your suggestions are totally valid so it would be great if you could open a PR with them so we can improve our tutorials. Alternatively I'll add this to our backlog but I'm not sure if we'll get spare time to handle this.

closed 2024-06-26 by marcoag