osrf/ros2_test_cases #1188: Cpp

marcoag, 2024-04-29:
Check the documentation for the 'Cpp' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Cpp page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Writing-an-Action-Server-Client/Cpp.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-09:
@marcoag i can check this.

fujitatomoya, 2024-05-09:
test result,

```bash
root@tomoyafujita:~/jazzy_ws# ros2 run custom_action_cpp fibonacci_action_server
[INFO] [1715228555.635159340] [fibonacci_action_server]: Received goal request with order 10
[INFO] [1715228555.635592671] [fibonacci_action_server]: Executing goal
[INFO] [1715228555.635805188] [fibonacci_action_server]: Publish feedback
[INFO] [1715228556.635854769] [fibonacci_action_server]: Publish feedback
[INFO] [1715228557.635965234] [fibonacci_action_server]: Publish feedback
[INFO] [1715228558.635816461] [fibonacci_action_server]: Publish feedback
[INFO] [1715228559.635957903] [fibonacci_action_server]: Publish feedback
[INFO] [1715228560.635845303] [fibonacci_action_server]: Publish feedback
[INFO] [1715228561.635938966] [fibonacci_action_server]: Publish feedback
[INFO] [1715228562.635956891] [fibonacci_action_server]: Publish feedback
[INFO] [1715228563.636035529] [fibonacci_action_server]: Publish feedback
[INFO] [1715228564.635983729] [fibonacci_action_server]: Goal succeeded
^C[INFO] [1715228566.957977614] [rclcpp]: signal_handler(signum=2)

root@tomoyafujita:~/jazzy_ws# ros2 run custom_action_cpp fibonacci_action_client
[INFO] [1715228555.634643624] [fibonacci_action_client]: Sending goal
[INFO] [1715228555.635468919] [fibonacci_action_client]: Goal accepted by server, waiting for result
[INFO] [1715228555.635950447] [fibonacci_action_client]: Next number in sequence received: 0 1 1 
[INFO] [1715228556.635993468] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 
[INFO] [1715228557.636355659] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 3 
[INFO] [1715228558.636011370] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 3 5 
[INFO] [1715228559.636290939] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 3 5 8 
[INFO] [1715228560.636003038] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 3 5 8 13 
[INFO] [1715228561.636268737] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 3 5 8 13 21 
[INFO] [1715228562.636339468] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 3 5 8 13 21 34 
[INFO] [1715228563.636428954] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 3 5 8 13 21 34 55 
[INFO] [1715228564.636220462] [fibonacci_action_client]: Result received: 0 1 1 2 3 5 8 13 21 34 55 
```

no problem observed, i think we can close this.

closed 2024-05-09 by ahcorde