osrf/ros2_test_cases #1189: Py

marcoag, 2024-04-29:
Check the documentation for the 'Py' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Py page](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Writing-an-Action-Server-Client/Py.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

Tacha-S, 2024-05-06:
:heavy_check_mark:  Step 1 - I was able to follow the documentation.
:heavy_check_mark:  Step 2 - The documentation seemed clear to me.
:heavy_check_mark:  Step 3 - The documentation didn't have any obvious errors.

The final result of the tutorial was as follows:

```bash
ubuntu@db0d2fcbaed3:~/ws$ python3 fibonacci_action_server.py
[INFO] [1714967932.750354062] [fibonacci_action_server]: Executing goal...
[INFO] [1714967932.750617817] [fibonacci_action_server]: Feedback: array('i', [0, 1, 1])
[INFO] [1714967933.752159796] [fibonacci_action_server]: Feedback: array('i', [0, 1, 1, 2])
[INFO] [1714967934.754092574] [fibonacci_action_server]: Feedback: array('i', [0, 1, 1, 2, 3])
[INFO] [1714967935.754943559] [fibonacci_action_server]: Feedback: array('i', [0, 1, 1, 2, 3, 5])
[INFO] [1714967936.756538558] [fibonacci_action_server]: Feedback: array('i', [0, 1, 1, 2, 3, 5, 8])
[INFO] [1714967937.758463194] [fibonacci_action_server]: Feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13])
[INFO] [1714967938.760176040] [fibonacci_action_server]: Feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21])
[INFO] [1714967939.762366535] [fibonacci_action_server]: Feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34])
[INFO] [1714967940.764826640] [fibonacci_action_server]: Feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])

ubuntu@db0d2fcbaed3:~/ws$ python3 fibonacci_action_client.py
[INFO] [1714967932.750373116] [fibonacci_action_client]: Goal accepted :)
[INFO] [1714967932.751154979] [fibonacci_action_client]: Received feedback: array('i', [0, 1, 1])
[INFO] [1714967933.754482211] [fibonacci_action_client]: Received feedback: array('i', [0, 1, 1, 2])
[INFO] [1714967934.756269580] [fibonacci_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3])
[INFO] [1714967935.755751334] [fibonacci_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5])
[INFO] [1714967936.758775031] [fibonacci_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8])
[INFO] [1714967937.760774158] [fibonacci_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13])
[INFO] [1714967938.761443842] [fibonacci_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21])
[INFO] [1714967939.764568102] [fibonacci_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34])
[INFO] [1714967940.767251989] [fibonacci_action_client]: Received feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
[INFO] [1714967941.767919185] [fibonacci_action_client]: Result: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
```

slowrunner, 2024-05-06:
I am working this ticket with configuration:
- Chip set: AMD64
- Build type: Debian
- OS: Ubuntu Noble 24.04
- DDS: FastDDS

Following the [Writing an Action Server/Client ](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Writing-an-Action-Server-Client/Py.html)

Issues for me:
1) The document does not mention running install/setup.bash, or local_setup.bash.  As a result, the instruction to send the goal reports:
```
$ ros2 action send_goal fibonacci custom_action_interfaces/action/Fibonacci "{order: 5}"
The passed action type is invalid
```
but executes fine after sourcing either local_setup.bash or setup.bash.

2) The doc does not say "edit fibanacci_action_server.py and add ```goal_handle.succeed()``` to the execute_callback()"
```
We can use the method [succeed()](http://docs.ros2.org/latest/api/rclpy/api/actions.html#rclpy.action.server.ServerGoalHandle.succeed) on the goal handle to indicate that the goal was successful:

    def execute_callback(self, goal_handle):
        self.get_logger().info('Executing goal...')
        goal_handle.succeed()
        result = Fibonacci.Result()
        return result
```

I need to continue later -

slowrunner, 2024-05-07:
4. Fourth issue for clarity:
```
Now let’s make our goal execution actually compute and return the requested Fibonacci sequence:
```
assumes the reader understands/recalls which file is being edited, which lines to add, and where.

Better would be:

Now let’s make our goal execution actually compute and return the requested Fibonacci sequence:
- edit fibonacci_action_server.py
- add the following lines to the execute_callback() method before the goal_handle.succeed() statement:
```
        sequence = [0, 1]

        for i in range(1, goal_handle.request.order):
            sequence.append(sequence[i] + sequence[i-1])
```
- add the line ```        result.sequence = sequence```  before the ```return result``` statement

to make the method be:

(existing execute_callback() block in doc)

5. Fifth issue for clarity - the doc states:
```
We’ll replace the sequence variable, and use a feedback message to store the sequence instead. After every update of the feedback message in the for-loop, we publish the feedback message and sleep for dramatic effect:

(then shows the entire changed file, without naming which file it is, leaving the learner to figure out which lines differ from their current copy of the file.)

```
While the doc could suggest simply replacing the entire file content with the boxed content, the point of the section is for the learner to understand specifically what is changed to provide feedback.  While the changes in the execute_callback() method are relatively easy to spot, the additional ```import time``` at the top of the file is easily missed.  If the user is not replacing the entire file.

Additionally, I prefer that the first import in any ROS 2 python program be the ```import rclpy``` and all rclpy related imports.  Maybe that is just a personal preference.
 
6) Sixth issue - in https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Writing-an-Action-Server-Client/Py.html#getting-a-result
```
Here’s the complete code for this example:
```
Better:
```
Here’s the complete code with getting the result by client node (fibanacci_action_client.py):
```

7) Seventh Doc issue:  in https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Writing-an-Action-Server-Client/Py.html#getting-feedback

```
Here’s the complete code for this example:
```

Better:
```
Here’s the complete code with getting the result and feedback to the client node (fibanacci_action_client.py):
```

So in summary:
- Step 1: Able to follow the doc:  yes
- Step 2: "Doc seemed clear": not as clear as I like.
- Step 3: "Doc no obvious errors": correct - no errors

I will not be upset if the doc issues I mention are not addressed - considering it all worked and I was able to succeed, no problem.

marcoag, 2024-05-08:
Nice work here @slowrunner. Since there's no obvious errors and you were able to follow I'm gonna go ahead and close this. It would be great if you can follow up with your corrections on a PR so we can discuss and improve the doc from there.

closed 2024-05-08 by marcoag

clalancette, 2024-05-10:
> 1. The document does not mention running install/setup.bash, or local_setup.bash.  As a result, the instruction to send the goal reports:

Understood, but in general in the "Intermediate" tutorials, we don't necessarily do this.  We could be exhaustive here, but the thinking is that if you've made it to these tutorials, you are familiar with the basics.

> 2. The doc does not say "edit fibanacci_action_server.py and add `goal_handle.succeed()` to the execute_callback()"

Yeah, I don't particularly love how this one uses colors to highlight the lines to change.  It is easy to miss, different from how all of our other tutorials do it, and may not be visible to color-blind people.

>     3. Third issue for me - why isn't this action server and client being developed in a package.  I don't want a server in my "home directory" or my ros2ws/ folder.

The thing is for Python that you don't *need* to use packages, and it is much more Pythonic to not use them.  We get complaints all the time that there is too much overhead for making ROS packages.  That is basically inevitable with C++ (since you need to setup the CMakeLists.txt and compile), but for Python we can just do scripts.  I guess you can also argue that since these are the "Intermediate" tutorials, people should already be familiar with packages, even for Python.  At the end of the day, I don't think either choice is wrong, so I'm inclined to leave this as-is.

> 4. assumes the reader understands/recalls which file is being edited, which lines to add, and where.

Yeah, agreed.  This is an extension of point 2.

> 5. While the doc could suggest simply replacing the entire file content with the boxed content, the point of the section is for the learner to understand specifically what is changed to provide feedback. While the changes in the execute_callback() method are relatively easy to spot, the additional import time at the top of the file is easily missed. If the user is not replacing the entire file.

Agreed.  An extension of point 2.

> 5. Additionally, I prefer that the first import in any ROS 2 python program be the `import rclpy` and all rclpy related imports.

This is not the style that flake8 enforces in our Python packages.  In particular, flake8 wants all Python stdlib imports to be first, so that is what we do everywhere.

> 6. Sixth issue - in https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Writing-an-Action-Server-Client/Py.html#getting-a-result

Agreed; we should clean that up a bit.

>     7. Seventh Doc issue:  in https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Writing-an-Action-Server-Client/Py.html#getting-feedback

Agreed, that is slightly nicer wording.

So I think the big things to fix here are the wording, and to somehow be more clear about which lines are changing.  I will attempt to make a PR to improve some of that in the next few working days.

Thanks for your thoughts!