osrf/ros2_test_cases #1192: Sane development environment

marcoag, 2024-04-29:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Process

    
            
### Test message generation using .msg, .srv, .action and .idl on top of the installed environment.
            
        

    
Generate messages from .msg, .srv, and .action files, which will automatically create their .idl files. Additionally, ensure that messages can be generated from .idl files alone.

Barry-Xu-2018, 2024-05-07:
OS: Ubuntu Noble 24.04 (via Docker)
Chip set: AMD64  (Intel i7-13700K)
Build type: Debian (Install ros-jazzy-ros-base package according to Jazzy manual https://docs.ros.org/en/jazzy/Installation/Ubuntu-Install-Debians.html)

Test way:
1. apt install ros-dev-tools
2. mkdir -p lazzy_src_ws/src && cd lazzy_src_ws/src
3. git clone https://github.com/ros2/example_interfaces.git
4. cd ../..
5. source /opt/ros/jazzy/setup.bash
6. colcon build --symlink-install --merge-install

Build is successful.  :white_check_mark:
```
root@7fd986e712c2:~/lazzy_src_ws# rm -rf build/ install/ log/
root@7fd986e712c2:~/lazzy_src_ws# colcon build --symlink-install --merge-install
Starting >>> example_interfaces
Finished <<< example_interfaces [11.4s]                     

Summary: 1 package finished [11.5s]
```
Check generated .idl by .msg, .srv and .action :white_check_mark:
```
root@7fd986e712c2:~/lazzy_src_ws# find build/ -iname '*.idl'
build/example_interfaces/rosidl_adapter/example_interfaces/msg/MultiArrayDimension.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/Int64MultiArray.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/Char.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/UInt32MultiArray.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/Int8.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/Empty.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/Int64.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/UInt16.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/Int32MultiArray.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/UInt32.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/Bool.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/Int16MultiArray.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/Float32.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/UInt64MultiArray.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/Int8MultiArray.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/Float64MultiArray.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/ByteMultiArray.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/UInt64.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/Float64.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/WString.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/MultiArrayLayout.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/Float32MultiArray.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/UInt8.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/UInt8MultiArray.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/Byte.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/Int16.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/Int32.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/String.idl
build/example_interfaces/rosidl_adapter/example_interfaces/msg/UInt16MultiArray.idl
build/example_interfaces/rosidl_adapter/example_interfaces/action/Fibonacci.idl
build/example_interfaces/rosidl_adapter/example_interfaces/srv/Trigger.idl
build/example_interfaces/rosidl_adapter/example_interfaces/srv/AddTwoInts.idl
build/example_interfaces/rosidl_adapter/example_interfaces/srv/SetBool.idl
```

Check generated code by .idl file :white_check_mark:
```
root@7fd986e712c2:~/lazzy_src_ws/build/example_interfaces/rosidl_typesupport_fastrtps_cpp# tree
.
└── example_interfaces
    ├── action
    │   └── detail
    │       ├── dds_fastrtps
    │       │   └── fibonacci__type_support.cpp
    │       └── fibonacci__rosidl_typesupport_fastrtps_cpp.hpp
    ├── msg
[... 48 lines cut ...]
    │   │   ├── multi_array_dimension__rosidl_typesupport_fastrtps_cpp.hpp
    │   │   ├── multi_array_layout__rosidl_typesupport_fastrtps_cpp.hpp
    │   │   ├── string__rosidl_typesupport_fastrtps_cpp.hpp
    │   │   ├── u_int16_multi_array__rosidl_typesupport_fastrtps_cpp.hpp
    │   │   ├── u_int16__rosidl_typesupport_fastrtps_cpp.hpp
    │   │   ├── u_int32_multi_array__rosidl_typesupport_fastrtps_cpp.hpp
    │   │   ├── u_int32__rosidl_typesupport_fastrtps_cpp.hpp
    │   │   ├── u_int64_multi_array__rosidl_typesupport_fastrtps_cpp.hpp
    │   │   ├── u_int64__rosidl_typesupport_fastrtps_cpp.hpp
    │   │   ├── u_int8_multi_array__rosidl_typesupport_fastrtps_cpp.hpp
    │   │   ├── u_int8__rosidl_typesupport_fastrtps_cpp.hpp
    │   │   └── w_string__rosidl_typesupport_fastrtps_cpp.hpp
    │   └── rosidl_typesupport_fastrtps_cpp__visibility_control.h
    └── srv
        └── detail
            ├── add_two_ints__rosidl_typesupport_fastrtps_cpp.hpp
            ├── dds_fastrtps
            │   ├── add_two_ints__type_support.cpp
            │   ├── set_bool__type_support.cpp
            │   └── trigger__type_support.cpp
            ├── set_bool__rosidl_typesupport_fastrtps_cpp.hpp
            └── trigger__rosidl_typesupport_fastrtps_cpp.hpp
```

Test passed. :white_check_mark:

Barry-Xu-2018, 2024-05-07:
Since there were no detailed testing instructions, I tested it in my own way.   
Please let me know if this is incorrect. Thanks！

closed 2024-05-07 by ahcorde