osrf/ros2_test_cases #1196: Demo Nodes (C++) executables

marcoag, 2024-04-29:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Links

* [Demo Nodes C++](https://github.com/ros2/demos/blob/jazzy/demo_nodes_cpp/README.md)

## Process

    
            
### The executables in the README are present and work as expected.

iam-shanmukha, 2024-05-04:
# Tested Demo Nodes in Docker. 
```
docker run -it -e DISPLAY=$DISPLAY -v /home/{USERNAME}/ros_workspaces/jazzy_ws:/home/ubuntu/jazzy_ws ghcr.io/sloretz/ros-testing:jazzy-desktop
```

## Noticeable updates are :  
- Parameter Events Node Name is `parameter_events_async` instead of `parameters_events_async`. You can see the output below. In the documentation it is mentioned as `parameters_events_async`
- Even Parameters Node Node name is `event_parameters_node` instead of `even_parameters_node`
- Rest all working as expected. 

## publisher and subscriber

```
root@vishnuk-Predator-PH315-54:/home/ubuntu/jazzy_ws# ros2 run demo_nodes_cpp talker
[INFO] [1714781728.523538365] [talker]: Publishing: 'Hello World: 1'
[INFO] [1714781729.523491003] [talker]: Publishing: 'Hello World: 2'
[INFO] [1714781730.523485955] [talker]: Publishing: 'Hello World: 3'
[INFO] [1714781731.523577677] [talker]: Publishing: 'Hello World: 4'
```

```
root@vishnuk-Predator-PH315-54:/# ros2 run demo_nodes_cpp listener
[INFO] [1714781728.523939070] [listener]: I heard: [Hello World: 1]
[INFO] [1714781729.523705785] [listener]: I heard: [Hello World: 2]
[INFO] [1714781730.523759486] [listener]: I heard: [Hello World: 3]
[INFO] [1714781731.523987879] [listener]: I heard: [Hello World: 4]
[INFO] [1714781732.523758057] [listener]: I heard: [Hello World: 5]
```

```
root@vishnuk-Predator-PH315-54:/# ros2 run demo_nodes_cpp listener_best_effort
[INFO] [1714781728.523952986] [listener]: I heard: [Hello World: 1]
[INFO] [1714781729.523699234] [listener]: I heard: [Hello World: 2]
[INFO] [1714781730.523761903] [listener]: I heard: [Hello World: 3]
[INFO] [1714781731.523782725] [listener]: I heard: [Hello World: 4]
[INFO] [1714781732.523737968] [listener]: I heard: [Hello World: 5]
[INFO] [1714781733.523855794] [listener]: I heard: [Hello World: 6]
```

## Server and Client

```
root@vishnuk-Predator-PH315-54:/# ros2 run demo_nodes_cpp add_two_ints_server
[INFO] [1714781910.500536430] [add_two_ints_server]: Incoming request
a: 2 b: 3
[INFO] [1714781910.936944718] [add_two_ints_server]: Incoming request
a: 2 b: 3
```

```
root@vishnuk-Predator-PH315-54:/# ros2 run demo_nodes_cpp add_two_ints_client
[INFO] [1714781910.500716544] [add_two_ints_client]: Result of add_two_ints: 5
root@vishnuk-Predator-PH315-54:/# ros2 run demo_nodes_cpp add_two_ints_client
[INFO] [1714781932.325223271] [add_two_ints_client]: Result of add_two_ints: 5
```

```
root@vishnuk-Predator-PH315-54:/# ros2 run demo_nodes_cpp add_two_ints_client_async
[INFO] [1714781909.532701060] [add_two_ints_client]: service not available, waiting again...
[INFO] [1714781910.937351535] [add_two_ints_client]: Result of add_two_ints: 5
root@vishnuk-Predator-PH315-54:/# ros2 run demo_nodes_cpp add_two_ints_client_async
[INFO] [1714781935.086246212] [add_two_ints_client]: Result of add_two_ints: 5
```

## On-oFF Timer

```
root@vishnuk-Predator-PH315-54:/# ros2 run demo_nodes_cpp one_off_timer

[INFO] [1714781994.230328170] [one_off_timer]: in periodic_timer callback
[... 521 lines cut ...]
[INFO] [1714784612.751044534] [loaned_message_talker]: Publishing: '4.000000'

```

## Matched Event Detect

```
root@vishnuk-Predator-PH315-54:/# ros2 run demo_nodes_cpp matched_event_detect
[INFO] [1714784666.523491609] [multi_sub_node]: Create a new subscription.
[INFO] [1714784666.524162323] [matched_event_detect_node]: First subscription is connected.
[INFO] [1714784666.524180097] [multi_sub_node]: Create a new subscription.
[INFO] [1714784666.524746366] [matched_event_detect_node]: The changed number of connected subscription is 1 and current number of connected subscription is 2.
[INFO] [1714784666.524760632] [multi_sub_node]: Destroy a subscription.
[INFO] [1714784666.525377337] [matched_event_detect_node]: The changed number of connected subscription is -1 and current number of connected subscription is 1.
[INFO] [1714784666.525391527] [multi_sub_node]: Destroy a subscription.
[INFO] [1714784666.525929061] [matched_event_detect_node]: Last subscription is disconnected.
[INFO] [1714784666.525941452] [multi_pub_node]: Create a new publisher.
[INFO] [1714784666.526485645] [matched_event_detect_node]: First publisher is connected.
[INFO] [1714784666.526498709] [multi_pub_node]: Create a new publisher.
[INFO] [1714784666.527058066] [matched_event_detect_node]: The changed number of connected publisher is 1 and current number of connected publisher is 2.
[INFO] [1714784666.527071641] [multi_pub_node]: Destroy a publisher.
[INFO] [1714784666.527672840] [matched_event_detect_node]: The changed number of connected publisher is -1 and current number of connected publisher is 1.
[INFO] [1714784666.527685748] [multi_pub_node]: Destroy a publisher.
[INFO] [1714784666.528199913] [matched_event_detect_node]: Last publisher is disconnected.
```

iam-shanmukha, 2024-05-05:
Hi @marcoag, can you please assign this to me?