osrf/ros2_test_cases #1198: Demo Nodes Native (C++) executables

marcoag, 2024-04-29:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Links

* [Demo Nodes C++ native](https://github.com/ros2/demos/blob/jazzy/demo_nodes_cpp_native/README.md)

## Process

    
            
### The executables in the README are present and work as expected.

mikaelarguedas, 2024-05-04:
Works as expected :heavy_check_mark: 

```
root@d8ff93e471d7:~/ros2_ws# ros2 run demo_nodes_cpp_native talker
[INFO] [1714865543.271983432] [talker_native]: eprosima::fastdds::dds::DomainParticipant * 0
[INFO] [1714865543.272549854] [talker_native]: eprosima::fastdds::dds::DataWriter * 0
[INFO] [1714865543.772266581] [talker_native]: Publishing: 'Hello World: 1'
[INFO] [1714865544.272218199] [talker_native]: Publishing: 'Hello World: 2'
[INFO] [1714865544.772186316] [talker_native]: Publishing: 'Hello World: 3'
[INFO] [1714865545.272218874] [talker_native]: Publishing: 'Hello World: 4'
[INFO] [1714865545.772256973] [talker_native]: Publishing: 'Hello World: 5'
[INFO] [1714865546.272240918] [talker_native]: Publishing: 'Hello World: 6'
[INFO] [1714865546.772209928] [talker_native]: Publishing: 'Hello World: 7'
[INFO] [1714865547.272198856] [talker_native]: Publishing: 'Hello World: 8'
[INFO] [1714865547.772285610] [talker_native]: Publishing: 'Hello World: 9'
[INFO] [1714865548.272205935] [talker_native]: Publishing: 'Hello World: 10'
```

iam-shanmukha, 2024-05-05:
Hi @marcoag , it looks like a duplicate of #1196. Can you please check.

closed 2024-05-06 by ahcorde