osrf/ros2_test_cases #1206: Intra-process executables

marcoag, 2024-04-29:
## Configuration

* DDS: FastDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04
* Chip set: AMD64

## Links

* [Intra-process demos](https://github.com/ros2/demos/blob/jazzy/intra_process_demo/README.md)

## Process

    
            
### The executables in the README are present and work as expected.

Ayush1285, 2024-05-02:
Hi @clalancette, I've tested this ticket. It is working as expected. Just a doubt, In the last step 5: "Image Pipeline With Two Image Views", I'm getting the expected output, that is the memory addresses of all the pointers are the same but values are fluctuating. I'm attaching the screen capture below.
[Screencast from 05-03-2024 01:03:43 AM.webm](https://github.com/osrf/ros2_test_cases/assets/64409716/54ca52c6-534d-42aa-9965-1e8c0ba9265e)

Ayush1285, 2024-05-02:
Terminal output for Step 1, Step 2
`root@ayushLegion:/# printenv ROS_DISTRO
jazzy
root@ayushLegion:/# ros2 run intra_process_demo two_node_pipeline
Published message with value: 0, and address: 0x55CADAA5D2A0
 Received message with value: 0, and address: 0x55CADAA5D2A0
Published message with value: 1, and address: 0x55CADAA5D2C0
 Received message with value: 1, and address: 0x55CADAA5D2C0
Published message with value: 2, and address: 0x55CADAA5D2A0
 Received message with value: 2, and address: 0x55CADAA5D2A0
^C[INFO] [1714677618.876459669] [rclcpp]: signal_handler(signum=2)
root@ayushLegion:/# ros2 run intra_process_demo cyclic_pipeline
Published first message with value:  42, and address: 0x55C7BAA32F50
Received message with value:         42, and address: 0x55C7BAA32F50
  sleeping for 1 second...
  done.
Incrementing and sending with value: 43, and address: 0x55C7BAA32F50
Received message with value:         43, and address: 0x55C7BAA32F50
  sleeping for 1 second...
  done.
Incrementing and sending with value: 44, and address: 0x55C7BAA32F50
Received message with value:         44, and address: 0x55C7BAA32F50
  sleeping for 1 second...
  done.
Incrementing and sending with value: 45, and address: 0x55C7BAA32F50
Received message with value:         45, and address: 0x55C7BAA32F50
  sleeping for 1 second...
  done.
Incrementing and sending with value: 46, and address: 0x55C7BAA32F50
Received message with value:         46, and address: 0x55C7BAA32F50
  sleeping for 1 second...
^C[INFO] [1714677657.832300535] [rclcpp]: signal_handler(signum=2)
root@ayushLegion:/# ros2 run intra_process_demo image_pipeline_all_in_one
[ WARN:0@0.045] global ./modules/videoio/src/cap_gstreamer.cpp (862) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
^C[INFO] [1714677788.912650558] [rclcpp]: signal_handler(signum=2)
root@ayushLegion:/# ros2 run intra_process_demo image_pipeline_with_two_image_view
[ WARN:0@0.035] global ./modules/videoio/src/cap_gstreamer.cpp (862) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
^C[INFO] [1714677869.015010977] [rclcpp]: signal_handler(signum=2)
root@ayushLegion:/#`

Ayush1285, 2024-05-02:
terminal output for Step 1 and Step 2:
[image]

clalancette, 2024-05-02:
> Hi @clalancette, I've tested this ticket. It is working as expected. Just a doubt, In the last step 5: "Image Pipeline With Two Image Views", I'm getting the expected output, that is the memory addresses of all the pointers are the same but values are fluctuating.

I think that is expected.

Thanks for testing!

closed 2024-05-02 by clalancette